budryerson/TFMini-Plus #1: Initial Run

mcradit, 2019-03-25:
Hello,

I just hooked a TF Mini Plus to a MKRWAN 1300. Pin 14 TX Pin 13 RX.
I am kind of a noob but how do I get the printf to work?  Will including ArdiunoSTL.h work?
Also I this error... 'bbTfmp' was not declared in this scope.

Any help would be greatly appreciated.

Thanks!

budryerson, 2019-03-26:
Sorry. The call to bbTfmp was incorrect. I have corrected the example code.
Download 'printf.h' from somewhere and include it in you sketch folder.
If you can wait until tomorrow, I will include 'printf.h' with the example code.
I'm not familiar with the MKR WAN 1300, but I understand that pins 13 & 14 are addressed as Serial1 so the example code should work for you.
Feel free to email me directly at < bud@budryerson.com >.
Remember - I'm new at this too.

mcradit, 2019-03-26:
Thanks for the quick response and for your work on this repo!
Now I think the issue is with the SAMD 21 micro controller and fprint.
Error: `'fdevopen' was not declared in this scope`
I will keep digging...

JBL99, 2019-04-11:
Hi @budryerson ,
I seem to be having the same issue as mcradit. When I try to compile TFMP_example, I get the same `'bbTfmp' was not declared in this scope` message on line 101 and 65. If I change this to tfmP as I understand your March 25 commit corrected, I then get the error that `'class TFMPlus' has no member named 'buildCommand' ` on line 101 and 82. 

In the comments of TFMPlus.cpp, it says that buildCommand() was changed to sendCommand(). If I change these, I then get the error `no matching function for call to 'TFMPlus::printStatus()' ` . I see this in the TFMPlus.cpp file, so I'm not sure why it isn't seeing this.

Are these issues just on my computer? I'm pretty new at this, so I'm trying to figure it out.

Also, it looks like you are using serial port 1. Are you using a Arduino Mega 2560? I'm trying to get this to work with three / four TFMini Plus sensors on a Arduino Nano. Will these work over software serial?

Thanks!

closed 2019-05-20 by budryerson