budryerson/TFMini-Plus #2: Frame rate command failed

paleeddie, 2019-05-16:
Hello,

I'm having an issue using your library / example sketch with a new TFMini Plus with firmware version 1.9.0, initially the frame rate command fails with the error data - Data: 5a 03 20 Status: CHECKSUM for example (the hex digits change every time, another example - Data: 5a 07 03 14 00 77 00) but always with a CHECKSUM status.

As the sketch continues to run, the library doesn't seem to get any data from the device, and instead reports the error - Loop:20 Data: 30 0d 0a 32 2e 30 30 0d 0a Status: SERIAL and loops forever with the Data values changing as I move objects in front of the sensor, but no distance measurements come through. 

I'm going to dig deeper into my issues tomorrow, but I thought I may as well see if you could offer any appreciated insight.

paleeddie, 2019-05-16:
Hello again,

I was able to un-comment the factory reset line in the sketch for one run and received data after that, however setting the framerate and saving the settings don't appear to be working with the following errors:

Lidar frame rate command failed. Data: 5a 07 03 14 00 77 00 Status: CHECKSUM
Lidar device settings not saved. Data: 5a 04 11 00 Status: CHECKSUM

budryerson, 2019-05-16:
Try this new version 1.3.3   I added a more robust header check on command replies. (It checks the first two bytes instead of just one) Also, it now completely clears the command reply buffer before reading.  There could have been a problem with residual data.  I hope this works better for you.  Let me know.  Thanks for your help.

paleeddie, 2019-05-20:
Thanks for the update, I tried the new library version and am now getting a much longer error message:

Lidar frame rate command failed. Data: 00 77 0a 40 09 9d 00 00 00 00 00 92 03 92 03 a0 03 00 00 0e 00 00 00 00 82 00 00 00 00 9e 02 00 00 00 00 7d cd 5b f7 b5 c3 18 d2 62 64 54 77 8d f3 5f 7f 13 ed f3 7b c5 3f 18 d2 6b 2a 7f af ed b6 2b f1 6f 7e 3a 6f f7 b2 fd de 32 ef 7f ad db f8 bf 4d 06 22 60 ee b3 8c b4 a9 31 6d d0 f2 3e e7 fa f5 7e bd af db 3f e3 d3 f4 d9 fd fe 79 56 a4 de f3 65 2d f5 6d Status: SERIAL

Not sure how much it matters (still pretty new at this) but I'm using an Arduino Uno with the software serial library and the TFMini Plus is hooked up through a logic level converter to bring the Arduino's 5V logic to 3V.

budryerson, 2019-05-21:
Did you comment out the factory reset line again, now that you are getting data?  The device takes a significant amount of time to accomplish a factory reset, so immediately subsequent commands may not work.

In that string of data, I don't see a single header byte.

paleeddie, 2019-05-21:
Thanks again for your replies and time, I found out the cable path I was using was the culprit, I had it setup with a breadboard between the Arduino and a logic level converter with long untwisted wires heading to the TFMini Plus. 

I'm now able to correctly set the frame rate, so thank you once more for all your help and a great library!

budryerson, 2019-05-21:
Happy to help. You made my day.

Oberjoa, 2019-06-07:
I have a similar issue: Not able to send any command, even using 5V logic level (and checking out the cables).
"Lidar frame rate command failed. Data: 00 0a 16 20 09 36 00 00 00 87 Status: SERIAL"
Another problem that i had is that SERIAL is defined in Arduino.h too. I changed "SERIAL" in "SERIALx" in the TFMPlus.h and TFMPlus.ccp files to avoid redifining.

JacquesGodin, 2020-01-07:
Hi,

could you share your code for sending these commands to the sensor? I've almost reached the end of the internet without understanding how I can send TFMini Plus commands without using Benewake's software

budryerson, 2020-01-07:
Hello Jaques,
   
  
  
   
  
  
   I will be happy to help.
  
  
   
    
   
   
    1) If you are successfully using Benewake's software, you are doing very well indeed.
   2) Have you looked into the library?  The code includes many useful comments.
  
  
   3) Are you using the TFMini Plus in serial (UART) or I2C mode?
   
  
  
   4) What exactly is your difficulty? What are you trying to do, and what is happening instead?
  
  
   5) Please include some of your code. Pictures are helpful, too.
   
  
  
   
  
  
   Regards,
   
  
  
   
  
  
   Bud Ryerson
   
  
  
   San Francisco

JacquesGodin, 2020-01-07:
Hey Bud,

thank you for the fast reply. Answers inline:

   _1) If you are successfully using Benewake's software, you are doing very well indeed._
        In fact I'm not. I cannot put the software to work, which is strange since using my python script I'm 
        able to (I'm using COM4 port for this. I've tried other ports, which work fine using pyserial but 
        don't work using the software)

   _2) Have you looked into the library?  The code includes many useful comments._
I did not have the chance yet, but I'll certainly do it.   

_3) Are you using the TFMini Plus in serial (UART) or I2C mode?_
I'm using a USB-to-TTL connection. Both in my Windows 7 PC and in my Raspberry Pi it works if used with the Python script

   _4) What exactly is your difficulty? What are you trying to do, and what is happening instead?_
   I'm trying to decrease the rate of Lidar readings. I need to make it a single reading per second, which is far from what I'm receiving now with the default definitions

5) Please include some of your code. Pictures are helpful, too.
Here's the code. I'm using a scheduler to get values every X seconds, but I've noticed that under certain conditions this does not work at all. I think it's because even with the scheduler giving just one value, the Lidar is sending more values which then confuse the processor. I believe I have to decrease the sensor frequency to the minimum possible in order to avoid this. I've learned that this can be done through the commands written in TFMini Plus product manual, yet it seems to work only in the Software.

```
import time
import serial
import schedule

ser = serial.Serial(
        port="COM1",
        baudrate = 115200,
        parity=serial.PARITY_NONE,
        stopbits=serial.STOPBITS_ONE,
        bytesize=serial.EIGHTBITS,
        timeout=1
)

def read_serial_port():
    ser.flushInput()
    x = ser.readline().decode("utf_8").rstrip('\n\r')
    y=float(x)
    print(y)

rate_in_seconds = 0.5
schedule.every(rate_in_seconds).seconds.do(read_serial_port)

while True:
    schedule.run_pending()
    time.sleep(1)
```

budryerson, 2020-01-08:
Jacques,
   
  
  
   
  
  
   Have you downloaded and installed the Benewake GUI on your Win7 PC?  Are you getting good data from your TTL to USB converter.  Are you able to send commands to the TFMini-Plus?
   
  
  
   
  
  
   Please be aware that the TFMini-Plus library I published is written in C++ for the Arduino microcontroller - not in Python for an Rpi.  I'll do what I can to help, but you might be asking the wrong person.
   
  
  
   
  
  
   That said, the default data frame rate of the TFMini-Plus is 100Hz, which is probably overrunning your serial input buffer.  You can to set the data frame rate of the TFMini-Plus to 2Hz (two times per second).  Or, you can set the device frame rate to 0Hz and then send a Trigger Detection command, which returns a single data frame, every 0.5 sec.,   Or, you could flush your serial input buffer every 0.5 sec and then read the next valid data frame.
   
  
  
   
  
  
   Let me know how that works, and good luck.
   
  
  
   
  
  
   Bud

JacquesGodin, 2020-01-08:
Hey Bud,

thank you for the tips. 

I successfully installed the Benewake GUI in my Win7 PC, yet I'm not getting any data from the Lidar. I thought it could be from the (1) TTL-USB converter or (2) the TFMini Plus itself, but the data I'm getting using the python script confirms that everything is working properly. Having that said, I cannot send any commands to the TFMini Plus through Benewake GUI.

This is the reason why I'm trying to find guidance to send commands to TFMini Plus through my python script.

Thank you for the detailed description. Are you able to do this change in the frame rate through your C++ script or is it through the Benewake GUI?

Oberjoa, 2020-01-09:
Sorry, but at the end i was not able to do this by code. Finally i settet
it up with an rs232 and the Benewakes software one time (i needed to set it
up to 500 measurements per second)

budryerson, 2020-01-10:
Jacques,

Yes, you can send a command to change the frame rate through the Benewake GUI.
The GUI is the easiest and most primitive way to communicate with the device.
You will need the GUI from time to time to verify that the device is working.
Please try again to use the GUI, and let me know the result.
Also, please send me a sample of the data from your Python script.

Thanks,

Bud Ryerson
San Francisco

JacquesGodin, 2020-01-11:
Hey Bud,

After contact Benewake's support, I finally could connect my TFMini to the GUI (I think I was using a outdated version of the software). Now, if I understood correctly, I should send commands to the TFMini via the GUI. I can see there's a 'Send command' button, but I have no clue on what's the command to use. In the user manual there a a few commands, but I'm not sure how to use it. Can you give me some clarity?

Also, once I succeeded in using the right command to define the default frequency to 1 reading per second, will the Lidar 'save' this configuration to be like this everytime?

Thanks,
J.

budryerson, 2020-01-13:
Jacques,

According to the Product Manual for the device,
the command format for setting the data frame-rate is:
   5A 06 03 LL HH SU

The 1st byte of every command is 0x5A
The 2nd byte is the number of bytes in the command sequence, in this case: 0x06
The 3rd is the command ID number, in this case: 0x03

The 4th and 5th bytes are the payload in "little endian" format.
That is the low order byte (LL) first, followed by the high order byte (HH).
For a data frame-rate of 1Hz, the payload would be 0x0100.
(A payload of 0x0001 would be 256Hz. And 1000Hz would be 0xE803.)

The last byte of the command is the check byte.  It is the low order byte
of the sum of all the previous bytes. In this case that's:
    0x5A + 0x06 + 0x03 + 0x01 x 0x00 = 0x0064

So, the command to set the data frame-rate to 1Hz is:
    5A 06 03 01 00 64

In order to take effect, this command must be followed
by the Save Settings command:
    5A 04 11 6F

I believe that a changed data-frame rate will survive a power reset;
but why don't you try it and you tell me.

Good hunting,

Bud

closed 2020-03-12 by budryerson