budryerson/TFMini-Plus #4: compile error

espresso0000, 2019-05-17:
exit status 1
'Serial2' was not declared in this scope

budryerson, 2019-05-17:
I was using an Arduino Mega2560. You can change the device serial port to Serial1 or use the Software Serial library.Sent via the Samsung Galaxy S7 edge, an AT&T 4G LTE smartphone

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/budryerson/TFMini-Plus/issues/4?email_source=notifications\u0026email_token=ABWD6ATYVBGAH3XJHFSYRCTPV4MTPA5CNFSM4HNYMSB2YY3PNVWWK3TUL52HS4DFUVEXG43VMWVGG33NNVSW45C7NFSM4GUPSQRA",
"url": "https://github.com/budryerson/TFMini-Plus/issues/4?email_source=notifications\u0026email_token=ABWD6ATYVBGAH3XJHFSYRCTPV4MTPA5CNFSM4HNYMSB2YY3PNVWWK3TUL52HS4DFUVEXG43VMWVGG33NNVSW45C7NFSM4GUPSQRA",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

espresso0000, 2019-05-18:
Did get that error just loading into arduino app and hitting compile before even getting to uploading to the arduino ATmega32u4 micro with the TFmini-Plus.

espresso0000, 2019-05-18:
Ok update, changing to Serial1 did allow it to compile, bout to test uploading it

espresso0000, 2019-05-18:
Using the micro I wired red to the 5v, black to the gnd, then put the Rx/Tx on pins then defined:
#include <SoftwareSerial.h>      // alternative software serial library
SoftwareSerial mySerial(10, 9); // RX, TX Pins

And output getting nothing:
Loop:06 Data: 00 00 00 00 00 00 00 00 00 Status: SERIAL
...

Thoughts, suggestions?

espresso0000, 2019-05-18:
Got it to work!

espresso0000, 2019-05-18:
And for fun even tried as:
        
        uint8_t tfDistE = uint8_t ( tfDist / 2.54);
        printf( " Distance: %04uin", tfDistE);  // Display the distance as inches.
        printf( " %04ucm", tfDist);             // Display the distance as centimeters.

espresso0000, 2019-05-19:
Do you think it is possible to run more then one TF on a single arduino?  I've done it with ultrasonic ping sensors, but when i attempted to use your setting for pins different then the TX/RX pins couldn't get it to take.

budryerson, 2019-05-21:
Yes, but not with the serial interface.  You would want to use the I2C interface for that purpose. This library does not support I2C yet.

cranefist, 2020-04-23:
I have the same issue, how did espesso0000 solve it?

If i just open the example, and try to verify it : "'Serial2' was not declared in this scope"

budryerson, 2020-04-23:
I'm happy to hear that your issue was resolved.
Does your device support I2C communication mode?  You might find that preferable to Serial.

Bud Ryerson
San Franciscso

ibrahimqazi, 2020-05-20:
@budryerson hello, I have a question, where should we put header file because when I compile it, it says `TFMPlus.h: No such file or directory`

budryerson, 2020-05-21:
Ibrahim:

**The Arduino IDE has libraries in multiple locations**

Arduino/AVR Libraries:
`C:\Program Files (x86)\Arduino\libraries`
`C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries
`

Contributed Libraries:
`C:\Users\[username]\Documents\Arduino\libraries`

Sketchbook Libraries:
This is determined by the sketchbook.path value in:
`C:\Users\[username]\AppData\Roaming\Arduino15\preferences.txt`

Or you can put the `TFMPlus` library folder in the same folder as your sketch.

For more information on installing libraries:
`http://arduino.cc/en/Guide/Libraries`

Please let me know whether that is helpful.

Bud Ryerson
San Francisco

ibrahimqazi, 2020-05-22:
@budryerson thank you for your reply. I tried all the options but unfortunately no success. Still the same error. :(

budryerson, 2020-05-22:
Okay.  We're going to fix this; but first I'll need more information.  Will you please create an archive of your project and relevant libraries and send it to me so I can duplicate the error? bud-at-budryerson-dot-com Thanks.

ibrahimqazi, 2020-05-22:
@budryerson I didn't make any changes, at the moment I just tried to compile your code, but it was not compiled.

budryerson, 2020-05-23:
I believe you.  I need more to go on.  Attach the sketch you are trying to run and the name of the folder you are running it from. Thanks, Bud

ibrahimqazi, 2020-05-25:
@budryerson hello, I have sent my archive with location details. Please check it.
Thank you

budryerson, 2020-05-25:
In the sketch `example.ino` on line #29, try replacing the brackets with quotes.
`#include "TFMPlus.h"  // Include TFMini Plus Library v1.3.4`
instead of  `#include <TFMPlus.h>`
This will look for the library header file in the current directory first, before trying system paths and IDE-defined locations.

ibrahimqazi, 2020-05-26:
Thank you so much. The code is finally compiled with renaming the object of SoftwareSerial class. Just want to know for my understanding, why it was unable to find the header files in those locations you told me. Although I put the files in those folders. Once again thnx

budryerson, 2020-05-26:
I am happy to hear that your project is advancing.  Did the quotation marks help?

I am puzzled that a compile error such as `TFMPlus.h: No such file or directory` was resolved "with renaming the object of SoftwareSerial class."

Also, I am concerned that your library placement may be inconsistent with good practice.  The locations you describe in `locations.txt` are difficult to reconcile with the archive you sent.  But further discussion of that here would be off-topic, I think.  We can continue by email, if you'd like.

As always, please let me know if I can be of any further help.

ibrahimqazi, 2020-05-27:
> I am happy to hear that your project is advancing. Did the quotation marks help?
>
> I am puzzled that a compile error such as `TFMPlus.h: No such file or directory` was resolved "with renaming the [...]

Yes the quotation mark solved my issue and I mentioned about "renaming the object of SoftwareSerial class" because you have written the code for another board may be Arduino Mega or something else but I am using UNO so that is why I said about renaming object. First I replaced <> with "" and then edited object of class SoftwareSerial. Once again thank you

closed 2020-07-06 by budryerson