budryerson/TFMini-Plus #6: Arduino freezed after new update

Ratthanin, 2019-07-16:
I used arduino nano + TFM mini plus + TVout library
I used serial to communicate with TFM.
it works fine and stable without trouble, but after new library updated.  
I recompile with the same code,
Arduino nano run normally when TFM distance = 0 cm (cannot get distance since low signal strength)
but whenever TFM can detect the distance,  arduino nano will stop and hang.
and it also make arduino unusable (it like the code was changed or deleted somehow)
I have to compile and upload code again to make it work.
(TFMplus connect to arduino nano hardware serial, this is the same serial that connect to  on board usb to serial converter for upload code.

Ratthanin, 2019-07-17:
I have tried changed code in many ways.  I found that the problem is about TVout library , not TFMminiplus.

budryerson, 2019-07-17:
Ratthanin,

I apologize for not replying sooner.
I am happy to hear that you have found the problem, and that it is not 
associated with the TFMiniPlus Library.
Please let me know if ever I can do anything to help.
Regards,

Bud Ryerson
San Francisco
< bud@budryerson.com >

On 17-Jul-19 1:49 AM, Ratthanin wrote:
>
> I have tried changed code in many ways. I found that the problem is 
> about TVout library , not TFMminiplus.
>
> —
>
On 16-Jul-19, Ratthanin wrote:

I used arduino nano + TFM mini plus + TVout library
I used serial to communicate with TFM.
it works fine and stable without trouble, but after new library updated.
I recompile with the same code,
Arduino nano run normally when TFM distance = 0 cm (cannot get distance 
since low signal strength)
but whenever TFM can detect the distance, arduino nano will stop and hang.
and it also make arduino unusable (it like the code was changed or 
deleted somehow)
I have to compile and upload code again to make it work.
(TFMplus connect to arduino nano hardware serial, this is the same 
serial that connect to on board usb to serial converter for upload code.

closed 2020-01-13 by budryerson