budryerson/TFMini-Plus #8: Do you only need one place to change the I2C address?

bpjdons, 2020-01-20:


budryerson, 2020-01-20:
You need to use the **TFMini-Plus-I2C** library to change the TFMini-Plus I2C address.  You may want to raise your issue in the repository for that library.

In the **TFMini-Plus-I2C** library, you will need to send a `SET_I2C_ADDRESS` command to change the I2C address, followed by a `SAVE_SETTINGS` command to save the change.  If you change the I2C address to one that is different from the default address (0x10), the library requires that you include the new, none-default address in every subsequent command.

I'm not sure I understood your question, so I hope this helps.

bpjdons, 2020-01-22:
Dear Bud,

 

Thanks for your information.

Can you also use the Arduino Nano for connecting to the TFMiniPlus.

If so, which pins do you have to connect to.

 

Thanks,

Bram Dons

bpjdons, 2020-01-23:
Hello Bud,

 

 

I have changed my TFMINI-Plus -with the SSCOM.5.12 program I got from Benewake support- to I2C and changed the address.

 

Right now I can’t connect the TFMINI within Ardupilot.

 

How can I change back to UART mode with the command SET_SERIAL_MODE?

 

What steps do I have to take with ARDUINO Nano to get this working.

 

After that I can hopefully change the I2C address with your program. (the Benewake SSCOM program does not seems to work properly.

 

Thanks in advance for your help.

 

Regards,

Bram Dons

budryerson, 2020-01-23:
Bram,

If you send a command to the TFMini-Plus to change from UART to I2C communication mode, all further communication must be in I2C format.  When the device is in I2C mode, you will not be able to use the SSCOM application.

However, while the device is powering up, it will be in serial mode for about ten seconds.  During that time, you might be able to reconnect with SSCOM and send the command `0x5A050A0069` to reset the device to UART  mode.

And yes, you can use an Arduino Nano to communicate with the TFMini-Plus in either UART or I2C mode.

Good luck,

Bud

bpjdons, 2020-01-23:
Bud,

 

Tried to reconnect with SSCOM but if failed on two of my TFMini-Plus.

 

I did put the command in the empty space, I guess that is allright?

 

Maybe the firmware could be corrupted because I have been playing around with the Arduino Nano.

 

I got the firmware form fengzuolei@benewake.com <mailto:fengzuolei@benewake.com>  but he could not inform me how to burn the firmware.

 

Any more suggestions?

 

Thanks,

 

Bram

 

Ps

The problem started with using the SSCOM to change the I2C address. After that I could connect to my Rangefinderx in Ardupilot.

This SSCOM application, I think is not working allright. Although Benewake supports tells you it is working perfect….

budryerson, 2020-01-23:
Bram,

I sincerely doubt that the firmware is corrupted or that SSCOM is not working correctly.

My suggestions are:
1) try using your Arduino Nano to communicate with the TFMini-Plus in I2C mode;
2) try to reach the device with SSCOM during the 10 second power up period;
(You didn't include the "0x" in the command `0x5A050A0069 ` you sent, did you?)
3) please, please try to communicate your problem clearly, precisely and thoroughly.  Include every step of your procedure and include all relevant code that you are using.  It will help you to think things through;
4) pictures of your setup and screen shots would be very helpful; and
5) never give up.

Bud

bpjdons, 2020-01-24:
Bud,

 

Tried to get the TFMini in UART mode, but it failed (see screenshot).

 

So I will try to connect with Arduino to the TFMini.

 

Althoug I am an IT professional myself I don’t have experiece with the Arduino environment. 

 

So could you advise me?

 

1.	Which connections do you use for connecting the TFMini to the Nano. 
2.	Do you use the TX/RX or the D2/D3 pins?
3.	Which Arduino program can I use to make the connection and how do I set this up?

 

Thanks in advance,

 

Regards,

Bram

budryerson, 2020-01-24:
Bram,
Show me how are you connecting SSCOM to the TFMini-Plus?
We are way off topic, so let's take this out of Github.
Write to me at: bud@budryerson.com
Bud

budryerson, 2020-04-24:
Bram was trying to setup three TFMini+'s to work together on a drone with a shared I2C bus.  He was having difficulty changing the I2C address and then maintaining communication with the devices.  Here are a couple of things we both learned:

1. Spaces are not allowed in Arduino sketch filenames.  Remember that!
2. While still in UART communication mode. you can command the device to change its I2C address   The default address of the TFMini+ is `0x10`.
3. The value of an I2C address cannot be checked in UART mode.  For that, you have to:
 a)  command the TFMini+ to change communication mode to I2C;
 b)  change your wiring to connect with the I2C bus pins; and
 c)   test the I2C bus for the presence of a device at that address (or better, check for all devices on the bus at any address.)
7. Thereafter, you can only communicate with the TFMini+ in I2C mode.  You can use the TFMini-Plus-I2C library for this purpose.  For any device not set to the default address of `0x10`, remember to include the correct I2C address with each command, including the command to `getData();`.
8. Sending a command to `Restore_Factory_Settings` will not change the communication mode back to UART.  However, it will change the I2C address of the TFMini+ back to its default value of `0x10`.

closed 2020-10-24 by budryerson