budryerson/TFMini-Plus #9: Wrong results

TrombiA, 2020-02-23:
Hello

Thank you for this library. I managed to make test program for Node MCU with Basic reading code and library. I can look distance results from serialmonitor and It gives mainly right results. However there are also few wrong results as you can see at picture. Where do you think those comes from and can I avoid somehow those wrong measurements?
[image]

budryerson, 2020-02-24:
Hello,
   
  
  
   
  
  
   I see that you are reading data 5x per sec (every 200ms). Is it posssible that the Frame Data-Rate is set faster than that, and that you are overrunning the serial buffer?  What is your Frame Data-Rate?
   
  
  
   
  
  
   
    I would like to know more about your use of the library with Node MCU.
   
   
    Please fill me in on all the details that you can.
   
   
    
   
   
    Best,
    
   
   
    
   
   
    Bud Ryerson
    
   
   
    San Francisco

TrombiA, 2020-02-25:
Hello 

I use default frame data rate 100hz and default serial baud rate 115200

I tried to change delay to 3ms and those wrong measurement values comes more rarely, something like once in a two minutes. So it looks like that buffer was overrunning like you guessed. Maybe my Node MCU board can't handle that speed either.

I read TFMPlus library text and noticed that there is commands which can be used for change default frame data rate. So can I just put that "define FRAME_1" to  my code beginning when I want to change that frame data rate to 1Hz? Or do I have to put some send command definition to code?

Like this (my code which I use for testing):

__________________________________________________________________________________________

#include <SoftwareSerial.h>
#include "TFMini.h"
**#define FRAME_1   //new line for change default frame rate to 1Hz??**

// Setup software serial port 
SoftwareSerial mySerial(D2, D1);      // Node MCU RX (TFMINI TX), Node MCU TX (TFMINI RX)
TFMini tfmini;

void setup() {
  // Step 1: Initialize hardware serial port (serial debug port)
  Serial.begin(115200);
  // wait for serial port to connect. Needed for native USB port only
  while (!Serial);
     
  Serial.println ("Initializing...");

  // Step 2: Initialize the data rate for the SoftwareSerial port
  mySerial.begin(TFMINI_BAUDRATE);

  // Step 3: Initialize the TF Mini sensor
  tfmini.begin(&mySerial);    
}

void loop() {
  // Take one TF Mini distance measurement
  uint16_t dist = tfmini.getDistance();
  uint16_t strength = tfmini.getRecentSignalStrength();

  // Display the measurement
  Serial.print(dist);
  Serial.print(" cm      sigstr: ");
  Serial.println(strength);

  // Wait some short time before taking the next measurement
  delay(200);  
} 
________________________________________________________________________________________________

I use Node MCU board (Lolin 0.1) and here is picture from my setup:

[image]

I also tryed this setup to use it with blynk app. Here is code for that:

_____________________________________________________________________________________

#define BLYNK_PRINT Serial    // Comment this out to disable prints and save space
#include <SPI.h>
#include <ESP8266WiFi.h>
#include <BlynkSimpleEsp8266.h>
#include <SimpleTimer.h>
#include <SoftwareSerial.h>   //Lisatty.22.2.-20
#include "TFMini.h"        //Lisattu.22.2.-20

// Setup software serial port alla olevat lisatty 22.2.-20
SoftwareSerial mySerial(D2, D1);      // Node MCU RX (TFMINI TX), Node MCU TX (TFMINI RX)
TFMini tfmini;

// You should get Auth Token in the Blynk App.
// Go to the Project Settings (nut icon).
char auth[] = "____________"; //Enter the Auth code which was send by Blink

// Your WiFi credentials.
// Set password to "" for open networks.
char ssid[] = "_______";  //Enter your WIFI Name
char pass[] = "______";  //Enter your WIFI Password

//DHT dht(DHTPIN, DHTTYPE);
SimpleTimer timer;

// This function sends Arduino's up time every second to Virtual Pin (5).
// In the app, Widget's reading frequency should be set to PUSH. This means
// that you define how often to send data to Blynk App.
void sendSensor()

  {
    float D = ((2.9*2.9*((900-(tfmini.getDistance()))/100))*0.55);  //D as distance
    float D2 = tfmini.getDistance();
    float S = tfmini.getRecentSignalStrength();
    Blynk.virtualWrite(V5, D );  //V5 is for Distance
    Blynk.virtualWrite(V6,D2 );
    Blynk.virtualWrite(V7,S );
  }
void setup()
{
  Serial.begin(115200);
  while (!Serial);
  Blynk.begin(auth, ssid, pass);

  // Step 1: Initialize hardware serial port (serial debug port)tama lisatty 22.2.-20
 
  // wait for serial port to connect. Needed for native USB port only
  

  // Step 2: Initialize the data rate for the SoftwareSerial port
  mySerial.begin(TFMINI_BAUDRATE);

  // Step 3: Initialize the TF Mini sensor
  tfmini.begin(&mySerial);    

  //dht.begin(); Poistettu kenoviioivlla 22.2.-20

  // Setup a function to be called every second
  timer.setInterval(10000L, sendSensor);
}

void loop()
{
  Blynk.run(); // Initiates Blynk
  timer.run(); // Initiates SimpleTimer
}

_____________________________________________________________________

It works fine with this code and I get values to blynk app. I were wondering that I can measure volume at storage with this TFmini lidar and blynk app. Thats why there are calculations with measurement results on blynk "D" value.

Br.
Tomi
Finland

budryerson, 2020-02-28:
Tomi,

It appears that you are calling the TFMini library header file ('TFMini.h').  You are NOT using the TFMini-Plus library.  There is no `getDistance()` function in the TFMini-Plus library.

Can that be part of the problem?  Please let me know.

Thanks,
Bud

closed 2020-03-12 by budryerson