budryerson/TFMini-Plus #10: Get firmware version failed.

ABartholsen, 2020-02-24:
Hi.

I am using a mega 2560, and am having some issues getting this up and running. I have tried connecting the tx and rx to pin 0 and 1, and I have also tried port 12 and 13. From what i can tell the mega 2560 does not communicate with the tfmini plus. Any idea what i am doing wrong?

My serial looks like this:

TFMPlus Library Example - 07JUN2019
Get firmware version failed. Attempt: 1 Data: Status: SERIAL

Get firmware version failed. Attempt: 2 Data: Status: SERIAL

Get firmware version failed. Attempt: 3 Data: Status: SERIAL

Lidar frame rate command failed. Data: Status: SERIAL
Loop:00 Data: 00 00 00 00 00 00 00 00 00 Status: SERIAL
Loop:00 Data: 00 00 00 00 00 00 00 00 00 Status: SERIAL
Loop:00 Data: 00 00 00 00 00 00 00 00 00 Status: SERIAL
Loop:00 Data: 00 00 00 00 00 00 00 00 00 Status: SERIAL
Loop:00 Data: 00 00 00 00 00 00 00 00 00 Status: SERIAL

Loop:01 Data: 00 00 00 00 00 00 00 00 00 Status: SERIAL
Loop:01 Data: 00 00 00 00 00 00 00 00 00 Status: SERIAL
Loop:01 Data: 00 00 00 00 00 00 00 00 00 Status: SERIAL
...

I am a bit new to this so any help would be much appreciated :)

budryerson, 2020-02-24:
Hi there,
   
  
  
   
  
  
   The TFMini-Plus Library example sketch is written to use Serial2 to communicate with the device.  On a Mega 2560, TX2 is pin 16 and RX2 is pin 17.  Of course, device TX connects to mega RX, and RX connects to TX.  But you knew that.
   
  
  
   
  
  
   Anyway, give it a try and let me know how it works.
   
  
  
   
  
  
   Bud Ryerson
   
  
  
   San Francisco

ABartholsen, 2020-02-25:
Hi. Thanks for the quick reply Bud Ryerson! 

For some reason I did not try RX2 and TX2 on the Mega board. Now I have tried them (16 and 17 as you said) without any different result. I was doing some more searching and came across the fact that the Mega uses 5V logic voltage in the serials while the Aruino Uno, and Arduino Due uses 3.3V logic level. And as I understand the TFMini Plus uses 3.3V as logic level. I have connected the TFMini Plus without any form of logic level converter. Could this be the problem? Have I perhaps fried the sensor by sending it the 5V signals? Or is the Mega not picking up the 3.3V signal from the sensor. I have an Arduino Uno I can try with the sensor later today to check if I can get it working that way. But I most say I was hoping to get it to work on the Mega as I plan on using two TFMini Plus - sensors.

Thanks again! :)
Arne Petter

ABartholsen, 2020-02-28:
I have now tested the sensors on an Arduino Uno, and turns out only one of the sensors was working. I guess the cause might be that I used the 5v serial signal directly on them. Maybe I tested one for a longer period of time, and thus it broke and not the other one.

budryerson, 2020-02-28:
Arne,
I use 5V signal levels and have never damaged one.  (I did kill one by applying 12V instead of 5V.) If the interior red light can be seen, the device is probably working and may be in the wrong mode.
Keep me posted.
Bud

ABartholsen, 2020-03-25:
The sensor did not give of the red light. So I must have broken it some how. I got a new one just now and it is working just fine. Thanks for your help Bud! :)

closed 2020-04-23 by budryerson