budryerson/TFMini-Plus #11: Junk lines when reading distance?

Stagematic, 2020-03-12:
Dear Bud,

I uploaded your example code to an Arduino UNO using software serial. Everything works, I omitted some lines, because I don't need them. 

The issue is, that when I'm reading distance, I get random CHECKSUM lines, which looks like these:

 Data: 59 59 ec 00 02 14 28 09 e4 Status: CHECKSUM
237
237
 Data: 59 59 ed 00 71 15 28 08 56 Status: CHECKSUM
237
237
237
236
237
 Data: 59 59 ed 00 0d 17 28 09 f3 Status: CHECKSUM
 Data: 59 59 ed 00 27 16 28 09 0c Status: CHECKSUM
237
237
237
 Data: 59 59 ec 00 3c 16 28 09 22 Status: CHECKSUM
 Data: 59 59 ec 00 45 16 28 09 2b Status: CHECKSUM
 Data: 59 59 ec 00 45 16 28 09 2b Status: CHECKSUM

What can I do to get clean distance readout? I need to pass raw distance data for the application that I'm using.

budryerson, 2020-03-12:
That looks bad.  The checksums are off by one.  I am not sure how that can happen. I will look into it.

budryerson, 2020-03-12:
That looks bad. The checksum bytes are off by one. The error appears to be in the data itself. I am not sure how that can happen.
Please let me know the baud rate and data-frame rate that you are using. Also, please check your voltage and current to the device.
Do you have another TFMini-Plus you can check it against?
Hmmmm.

budryerson, 2020-03-14:
I see in the example code that the program loop is running at about 100Hz; but the data frame-rate of the device is being set to 20Hz.  That appears to be incorrect.  In the file `TFMP_example.ino` please change line #126 from `framerate(20);` to `framerate(100);`

I hope that helps to minimize the errors you are getting.  And I will change the example code.

Thank you,

Bud Ryerson
San Francisco

budryerson, 2020-07-09:
Someone reported a similar issue because they deleted the `#include "printf.h"` statement from the `TFMP_example.ino` sketch.

You wrote: _"I omitted some lines, because I don't need them."_  Please let me know which lines you decide to omit.

Thanks,

Bud Ryerson
San Francisco

SantiFerre, 2020-11-26:
I had same problem with TFMini-s. I solved it using hardware serial instead of the SoftwareSerial library.

Thanks for the library! =)

Snipersdimages, 2021-03-13:
Can i use I/O

Snipersdimages, 2021-03-13:
With an Arduino uno, why not use I / O communications to avoid problems with the SoftwareSerial library?

budryerson, 2021-03-13:
The so-called I/O mode is a simple binary output from the device that indicates whether an object is at or within a certain range of distances.  To me, that's not very fun or useful; and **this library does not support I/O Mode**.  Or maybe you meant something else?  Please let me know if I answered you question.
Thanks,
Bud Ryerson
San Francisco

Choikyungho9, 2021-03-17:
I have the same problem.

`Data: 59 59 ec 00 3c 16 28 09 22 Status: CHECKSUM`

How do I delete this line?

Why is this line appearing in the output??

Snipersdimages, 2021-04-24:
Bonjour,

J'utilise un TFMini pour détecter la présence d'un animal à un endroit
précis et déclencher un APN. Une simple sortie binaire me serait nécessaire
et suffisante, mais je ne sais pas comment la programmer et configurer la
distance et la plage de détection.
L'avantage serait de ne pas utiliser les bornes séries et d'utiliser
directement un microcontrôleur, moins gourmant en énergie qu'un Arduino.

La traduction avec Google est meilleur que ce que je pourrais faire !

Cordialement

J.L. Cagnet
Trèlévern France

budryerson, 2021-04-24:
J.L.

You wrote, "J'utilise un TFMini pour détecter la présence d'un animal à un endroit précis et déclencher un APN. Une simple sortie binaire me serait nécessaire et suffisante."  That is true, and detecting the presence of an animal is a perfect use of the I/O mode. This library does not support I/O Mode, and if you are not using an Arduino microcontroller ("L'avantage serait de ne pas utiliser les bornes séries et d'utiliser directement un microcontrôleur"), then there is no use for this library.

Please read the manufacturer's Product Manual for the device, and then use the manufacturer's GUI to set the device to I/O mode.  If I can be of any further help, or answer any questions, please let me know.

Sincerely,

Bud Ryerson
San Francisco

Snipersdimages, 2021-04-25:
Bonjour,
Je vous remercie sincèrement  pour votre réponse.

J'ai déchiffré le manuel anglais du fabricant et je ne sais pas ce qu'est
l'interface graphique.
Je ne suis pas certain de la manière de régler le mode E/S et si je
pourrais ensuite revenir facilement en mode UART.
Je suis un photographe animalier amateur qui souhaite développer un produit
introuvable dans le commerce. Je maitrise mal le vocabulaire d'ingénierie
électronique (et autres). Je ne peux comprendre et adapter l'utilisation du
TFmini, qu'avec des exemple concret.
Je continu donc ma quête d'un utilisateur du TFmini en mode I/O. En
attendant je retourne à une utilisation d'un simple capteur de mouvement
infra rouge, mais qui ne correspond pas à mes objectifs.

Cordialement

M. Cagnet

budryerson, 2021-04-25:
My dear M,

The GUI (Graphical User Interface) application is freely available from the manufacturer; but you will need an additional device, a USB2TTL (USB to serial) adapter between your computer's USB port and the serial interface lines of the TFMini-Plus.  The adapter is inexpensive but essential.  You must have one to check if your TFMini-Plus is working, if for no other reason.

In the end, I do not understand how you will be able to _"développer un produit introuvable dans le commerce"_ if you do not understand _"le vocabulaire d'ingénierie électronique (et autres)."_  Forgive me, but his stuff does not happen by magic.  You will need to try to understand, experiment, fail, try to understand more,  experiment again, fail again, and never, ever give up.  We are all here to help you, but you have to try.

[This](https://www.amazon.com/HiLetgo-CP2102-Converter-Adapter-Downloader/dp/B00LODGRV8/) might work for you.  Let me know when it arrives.

All the best,

Bud Ryerson
San Francisco

Snipersdimages, 2021-04-26:
Bonjour,
Merci pour votre réponse.

Lors de l'achat du TFmini, j'avais acheté un module USB to TTL et
télécharger l'application graphique, mais je ne les ai pas utiliser puisque
je peux communiquer avec un Arduino.
Je pensais que l'Arduino pouvait remplacer le module USB to TTL et trouver
une bibliothèque me permettant de configurer mon TFmini en E/S.  Il semble
donc que les codes de configuration de Benewake ne puisse être adressé au
TFmini, que par l'interface Graphique. Je vais donc suivre vos conseils dès
que j'aurais trouvé le nom des variables qui  me permettent de régler la
zone de détection du TFmini.
Actuellement j'enregistre la distance sur une cible avec un simple BP. Le
déclenchement se fait à cette distance, mais c'est trop précis, J'ai
rajouté une zone mini maxi, correspondant à la profondeur de champs de
l'APN. J'ai donc "recréer" le fonctionnement en E/S sur le port série avec
mes propres variables.

Ma prochaine étape et de réduire drastiquement la consommation...
Je vais suivre vos encouragements "d'essayer de comprendre, d'expérimenter,
d'échouer, d'essayer de mieux comprendre, d'expérimenter à nouveau,
d'échouer à nouveau et de ne jamais abandonner." et vous informerais de mes
progrès, mais la photo animalière est une activité extrêmement chronophage.

Et merci encore de m'avoir consacré un peu de votre temps !

M. Cagnet

budryerson, 2021-04-26:
Maybe this will help.  All data bytes are in hexadecimal.

The I/O output mode is enabled and disabled by this 9 byte command:
5A  09  3B MODE DL  DH  ZL  ZH  SU
```
Command byte number:
0  5A:  Header byte, starts every command frame
1  09:  Command length, number of bytes in command frame
2  3B:  Command number

3  MODE: 0--IO Mode OFF, standard data mode
         1--IO Mode ON, output: near = high and far = low
         2--IO Mode ON, output: near = low and far = high

4  DL: Near distance lo order byte of 16 bit integer
5  DH: Near distance hi order byte

6  ZL: Zone width lo byte
7  ZL: Zone width hi byte

8  SU: Checkbyte (the lo order byte of the sum of all the other bytes in the frame)
```
If an object's distance is greater than the Near distance (D) plus the Zone width (Z) then the object is "far"
If the distance is less than the Near distance (D) then the object is "near".
The Zone is a neutral area.  Object distances measured in this range do not change the output.
The output can be set to be either high when the object is near and far when it's low (Mode 1) or low when it's near and high when it's far (Mode 2).
The high level is 3.3V, the low level is 0V.

Snipersdimages, 2021-04-26:
Merci ! Merci ! Merci ! c'est exactement ce que je cherchais !

Soit je n'ai pas trouvé ces explications, soit je n'ai pas compris la
documentation anglaise.

Respect !

M. Cagnet

Snipersdimages, 2021-04-26:
Donc si j’ai bien compris, La sortie I/O n’est pas utilisable pour mon
détecteur de faune, puisque le changement d’état ne se ferait qu’a la
sortie de la zone de prise de vue, voir pas du tout pour un animal qui la
traverserait de droite à gauche.

Progresse je ?

Snipersdimages, 2021-04-26:
La solution de réduire la zone à zéro permettrait de détecter un passage mais également le passage traversant proche.
Mon petit programme fait mieux que ça !

closed 2022-04-04 by budryerson