budryerson/TFMini-Plus #13: Converting tfDist into meters

andyk735, 2020-04-04:
Hi Bud, 

Using your example code, I would like to convert the tfDist value, which is outputted in centimeters, into meters. I have tried performing math using a double: 
`if ( tfmP.getData( tfDist, tfFlux, tfTemp)) // Get data from the device.
    {
      double tfMeter;
      tfMeter = tfDist / 100.0;
      printf( " Dist: %f", tfMeter);         // Display the distance.`

However, it always outputs something like:

`Loop:01 Dist: ?
Loop:02 Dist: ?
Loop:03 Dist: ?
Loop:04 Dist: ?
Loop:05 Dist: ?`

Do you have any idea of how I could solve this issue? I am not sure, it may be because tfDist is a uint16_t integer type, but I am not sure how I would fix it regardless. Hope you can help, thank you.

budryerson, 2020-04-05:
Hey AndyK,
Try recasting tfDist as a floating point number:  `tfMeter = (float)tfDist / 100.0;`
Let me know what happens.
Bud

andyk735, 2020-04-17:
Yep, it worked! Thank sorry for not responding sooner. 

I ran into another issue though. I am using an Arduino Uno and am trying to use the sensor with a button as well. I use Software Serial and included mySerial to get the TFMini-Plus working, because I did not know how to get it to work any other way. However, for some reason, this makes my button act weirdly, making it simply not act as a button. It's either always on or off. 
Any help?

budryerson, 2020-04-17:
Andy,
Your question is too vague for me to answer.  Please provide a more thorough explanation of the problem.  A sample of your code would be helpful.  And pictures or a wiring diagram, too.
Bud

closed 2020-07-09 by budryerson