budryerson/TFMini-Plus #14: Another "Get Firmware Version" failed

mprowe, 2020-04-17:
I am running your example on an ESP32 and the **tfmP.getData( tfDist, tfFlux, tfTemp)** is working (although I have doubts about the Temp, which I'll skip over for now).
However, I can't get any output from the  firmwareVersion() function. Run as-is, the debug output is:
```
FMPlus Library Example - 07JUN2019
 Status: SERIAL
Get firmware version failed. Attempt: 1 Data:
 Status: SERIAL
Get firmware version failed. Attempt: 2 Data:
 Status: SERIAL
Get firmware version failed. Attempt: 3 Data:

```
...and re-running with **tfmP.printStatus(true);**, `gives:
```
FMPlus Library Example - 07JUN2019
 Status: SERIAL
Get firmware version failed. Attempt: 1 Data: 00 00 00 00 00 00 00 00 00
 Status: SERIAL
Get firmware version failed. Attempt: 2 Data: 00 00 00 00 00 00 00 00 00
 Status: SERIAL
Get firmware version failed. Attempt: 3 Data: 00 00 00 00 00 00 00 00 00
```
A quick look through TFMPlus.h, throws up an immediate question. That header has this:
**#define    OBTAIN_FIRMWARE_VERSION    0x00010407   // returns 3 byte firmware version**
...but looking through the TFmini-Plus-T-01 A04_Product Manual.pdf which you have very helpfully included in the library, Table 11 says that the Obtain firmware version is **5A 04 01 5F**
Should they be the same?
Regards,

budryerson, 2020-04-17:
The chip temperature data always seemed a little hot to me.  Is that what you mean?
This library uses defined commands derived from the Product Manual according to the format:
```
// This library's 'sendCommand( cmnd, param)' function
// defines a command (cmnd) in the the following format:
// 0x     00       00       00       00
//     one byte  command  command   reply
//     payload   number   length    length
```
The Firmware command is defined as `0x0147`
  0 - payload,
  1 - command number,
  4 - command length in bytes,  and 
  7 - reply length expected in bytes

At the top of the `sendCommand()` code, a proper array of command data is created from this four byte code.

(Should they be the same?  Some libraries have a separate block of code for each command.  I chose to write one block of code that interprets each command from a defined set.  Your mileage may vary.)

To continue... at line #251-256 of the `sendCommand()` code in `TFMPlus.cpp`, a firmware version number is plucked from the 7 bytes of data returned by the Firmware command in little endian order, i.e. the first byte is the least significant.  As you see in the product manual at the top of page 18: Bytes V1 V2 V3 are reversed to become Version V3.2.1.

You write that you are getting good serial data from `getData()` but that you are not getting any data at all from the Firmware command.  Did the example's following two commands:
```
frameRate( 100);   // set device data frame-rate to 100Hz
saveSettings();      // save the frame-rate setting
```
return an error status as well?  Please let me know.  We will fix this thing.

Bud Ryerson
San Francisco

mprowe, 2020-04-17:
Hi Bud,
Thank you for the very rapid response. Yes, they do return errors. Forgive me for not mentioning, I wanted to deal with one issue at a time. Anyway, this is the start of my **setup()** section:```

void setup()
{
    Serial.begin( 115200);   // Intialize terminal serial port
    delay(20);               // Give port time to initalize
    Serial << endl << "FMPlus Library Example - 07JUN2019" << endl;

    Serial_1.begin(115200, SERIAL_8N1, 34, 35);   //baud, mode, Rx, Tx (connected to TFMiniPlus)
    delay(200);               // Give port time to initalize
    tfmP.begin( &Serial_1);   // Initialize device library object and...
    delay(500);               // pass device serial port to the object.
    
    init_oled();
    
    // Send commands to device during setup.
    firmwareVersion();  
    frameRate(100);
    saveSettings();
    //factoryReset();
    while (1) yield();    //Stop here. To grab output on terminal
```
... and my output is this:
```
FMPlus Library Example - 07JUN2019
 Status: SERIAL
Get firmware version failed. Attempt: 1 Data:

 Status: SERIAL
Get firmware version failed. Attempt: 2 Data:

 Status: SERIAL
Get firmware version failed. Attempt: 3 Data:

Lidar frame rate command failed. Data: 00 41 08 f0 08 95 00 00 00 00 e1 01 00 00 b0 10 40 3f 00 00 00 00 e8 03 00 00 00 00 00 00 Status: SERIAL
 01 00 00 00 64 ec fb 3f e4 10 40 3f 00 00 00 00 14 10 40 3f 00 00 00 00 e8 03 00 00 00 00 00 00 01 ff ffLidar device settings n Status: SERIAL

```
Regards, Martin

budryerson, 2020-04-17:
Martin,
Wow! That is just garbage.  I do not see any HEADER bytes.
Are you sure you're getting good distance, flux and temperature data?
Please confirm that the device is working correctly with the manufacturer's GUI.
Thanks,
Bud

mprowe, 2020-04-18:
Bud,
Okay, one by one...

1. Am I getting good dist, flux & temp? Yes. As far as I can tell - temp is 20degress high though. 
2. Manufactures GUI? Yes. See attached screenshot.
[image]
3. This screenshot is the raw data from the tfMini Plus as seen with CuteCom (in Hex display mode). Is that any help?
[image]

One more point I perhaps should own up to? Is that I could not get your **printf()** to work? So all the Serial output is being done with Streaming library [https://github.com/janelia-arduino/Streaming](url)

Regards, Martin

mprowe, 2020-04-18:
Hi Bud,
A quick update on today's work. Went back to basics. I don't have an Arduino Uno but I do have a collection of ATMega328xx. First, on an ATMega328**P** the only modification to your TFMP_example was to un-Rem lines 32 & 33.
[image]
This runs as you would expect, the problem is: That SoftwareSerial is not that reliable at 115200baud and there are aprox 40% checksum errors in the **getData()** loop. This I'm not surprised at. I think 9600 is the recommended max for the SoftwareSerial library.
Next, I rebuilt the breadboad with an ATMega328**PB** which as you know has 2 hardware UARTs Serial0 (or just Serial for compatibility) on pins PD0 & PD1 and Serial1 on pins PB3 & PB4. Both of which are able to run safely at 115200. Re-Reming lines 32 & 32 to remove the SoftwareSerial and renaming second serial port names at lines 119 & 121:
[image]
Now, the example runs cleanly. No checksum errors. It displays the version number  (v2.0.9) and is able to set the Frame Rate (I note that you still have it set at 20 in your library example - you mentioned that you were going to change it to 100 elsewhere in this forum).

Thus, it seems to me, that there is an unexpected difference in behaviour between your program when compiled for an ATMega328 and when compiled for an ESP32.

I have tried to follow your code into the **getData()** and **sendCommand()** to try and figure it out. But my coding is not up to it - all that C++ indirection just leaves my head swimming! If you could offer me any tips on how to debug this minor problem, they would be most welcome. I would really like the TFMini Plus to run on the ESP32 for the network support it offers.

Regards, Martin

budryerson, 2020-04-18:
Martin,

I assume that you mean "20 degress high" relative to room temperature.  I get that, too.  Maybe because it's measuring at the chip.

Thank you for demonstrating that your device is working properly.  I see good data in the hex dump, specifically the the two header bytes `0x5959` followed by six bytes of data and a checksum byte.  Perfect.

I'm sorry to hear about **printf()**.  It is irregular; but I find it to be very common.  I'm surprised that it did not work.  Did you include the `printf.h` file in your sketch folder?

Thanks for the link to Streaming.  I'm looking into that. Maybe it will replace **printf()**.

I regret the error in the example that sets the data frame-rate to 20Hz and then sets the loop delay to 10 msec.  The loop delay should be 50 msec. or the fame-rate at 100Hz to match.  The most recent copy of the example corrects that.

Using an Arduino ProMini, I have not experienced any problems with **SoftwareSerial** at 115200.   But I see that the ESP32 has 3 UART interfaces (the USB connector and two more).  If we _must_ use **SoftwareSerial** then I see there's an **EspSoftwareSerial** library.  I also see that the ESP32 has two I2C interfaces.  In my opinion, UART is very messy.  You may prefer using I2C communication with your TFMini-Plus.

In order to properly duplicate your problem, I'll need to acquire an ESP32 device (only $14 for two on Amazon!).  But in the meantime, please send me **\*all\*** the code that you are using and I'll do everything I can to help figure out what's going on.  Thank you for your patience.

Tally ho,

Bud

budryerson, 2020-04-21:
Martin-
Two MELIFE ESP-32S Development Boards arrived today.  I look forward to trying them out.
Please send me the code you are having trouble with.  Hopefully, we can duplicate the problem.
I'll let you know what happens.
Bud
[image]

mprowe, 2020-04-21:
Sorry for the silence Bud! I had made a long reply to your last over and sent it via a reply eMail to the github notification rather than here. Silly me...
Anyway, rather than go over the ground again, let me keep it simple and just post my code for you to have a look at.
One additional bit of information: I get the same problem on the ESP8266. Not surprising really considering the Espressif lineage.

Regards, ~Martin
[Lidar_Client.zip](https://github.com/budryerson/TFMini-Plus/files/4511962/Lidar_Client.zip)

budryerson, 2020-04-21:
Martin,

My short answer at this point is that `HardwareSerial Serial_1(1)` is receiving correctly; but it is not transmitting correctly. I would write a little serial transmit sketch, hook my ESP32 Serial1 port up to CuteCom or whatever and make sure of what the damn thing is spitting out data correctly.

My longer answer is that your sketch, `Lidar_Client.ino`, does not include `Arduino.h`.
However you do include `Streaming.h`, which does include `Arduino.h`, so there you go.
`Arduino.h` includes the usual and necessary: `stdlib.h`, `stdint.h`, `stdbool.h`, `stddef.h`, `stdarg.h`, `stdio.h`, `string.h`, and `math.h`. But `Arduino.h` also includes: `stdlib_noniso.h`, `binary.h`, `esp8266_peri.h`, `twi.h`, `core_esp8266_features.h`, and `core_esp8266_version.h`.  I thought the **esp8266** stuff was interesting.

I don't believe that you told me about including `HardwareSerial.h`.  I don't understand that. Hardware serial ports are available as **Serial1**, **Serial2**, etc, aren't they?

`HardwareSerial.h` includes `Stream.h`, and `Stream.h` includes `Print.h`, and `Print.h` includes `stdint.h`, `stddef.h`, `WString.h`, `Printable.h`, `stdlib_noniso.h`... and at that point I just lost it.

Please explain to me briefly why `HardwareSerial.h` is necessary.

As I said, my working assumption is that your **Serial_1** transmitter is not working correctly. I would try writing a sketch without Streaming.h and HardwareSerial and see how that behaves.

In the meantime, I'll try to get my shiny new **Esp32** fired up.

I'm sorry about the trouble. All the best,

Bud

mprowe, 2020-04-22:
Bud,
Some quick answers while I check-out some of the other questions you have asked:
"Please explain _to me briefly why HardwareSerial.h is necessary?"_

As always, I am working from examples. And when at first **Serial1.begin(115200)** was not working, several post like [this](https://quadmeup.com/arduino-esp32-and-3-hardware-serial-ports/) and [this](http://www.iotsharing.com/2017/05/how-to-use-serial-arduino-esp32-print-debug.html) suggested that the **HardwareSerial.h** route was the solution.
At first sight, anyway, it seemed to work.... 
More answers to follow...
~Martin

mprowe, 2020-04-22:
Mia Culpa... **Mia Culpa...** So sorry to have troubled you it is all my fault!  

For several reasons, I could not use the default pins (GPIO16 & GPIO17) for Serial1. So, excited by the prospect that I could pick alternatives due to the ESP32 multiplexed GPIO feature, I went for a more convenient (for me) pair.
What I had completely overlooked was that some GPIOs are more GPIO'ish than others.
[image]

You are also correct in that I don't need the <HardwareSerial.h> includes. The only chage (I think) to basic Arduino Serial coding style is an extension to the Serial.begin() to cater for the option to choose alternative GPIO pins: **Serial.begin(baud, format, RX, TX)** 
**Important:** don't use GPIO34, 35, 36 or 39 for the TX pin!

Thanks for all your help Bud. I think I'm up 'n running now...

~Martin

closed 2020-04-22 by mprowe

dhruvamp, 2020-10-27:
hey guys,
here this code works very well on ESP32 using two pwm pins as a hardware serial.
hope you may find it worthy.

/*  TFMINI PLUS LIBRARY > "TFMPlus.h"

    TFMINI PLUSE SENSOR :: ESP 32 PIN CONNECTION
                   RED  ::  +5V
                 BLACK  ::  GND
                 GREEN  ::  RX PINS (16,22)
                 WHITE  ::  TX PINS (17,23)//NO NEED TO CONNECT THESE PINS

    HARDWARE SERIAL 1 & 2 ARE CONFIGURED FOR EACH SENSOR
    CREATED TWO TFMPlus PROJECT FOR EACH LIDAR SENSOR

    tfmP.getData( tfDist, tfFlux, tfTemp)
    tfDist = GETS DISTSNCE VALUE IN CM FROM THE TFNPlus PROJECT //cms TRAVELLED BY LASOR
    tfFlux = GETS LASOR STRENGTH VALUE FROM TFMPlus PROJECT //SIGNAL STRENGTH
    tfTemp = GETS TEMPERATURE VALUE FROM TFMPlus PROJECT //ESP32 CORE TEMPERATURE

    HERE, WE HAVE USED tfmP.getData( tfDist) TO GET ONLY DISTANCE IN cm

*/

//ADD LIBRARIES FOR HARDWARE SERIAL AND TFMINI PLUS
#include <HardwareSerial.h>
#include <TFMPlus.h>  // Include TFMini Plus Library v1.4.1

//CREATE TWO PROJECT FOR TWO SENSORS IN TFMPlus
TFMPlus sensor1project;
TFMPlus sensor2project;

//DEFINE HARDWARE SERIALS OUT OF 0, 1, 2.
HardwareSerial sensor1serial(1);
HardwareSerial sensor2serial(2);

int16_t tfDist = 0; //SET INITIAL VALUE O
unsigned long int t1 = 0, t2 = 0, T = 0,velocity;
int dist1 = 0 , dist2 = 0;

void setup()
{
  Serial.begin( 115200);

  sensor1serial.begin(115200, SERIAL_8N1, 23, 17);
  sensor2serial.begin(115200, SERIAL_8N1, 22, 16);

  sensor1project.begin( &sensor1serial);
  sensor2project.begin( &sensor2serial);
}

void loop()
{

  while (sensor1serial.available() > 0)
  {
    if ( sensor1project.getData( tfDist)) // Get data from the device.
    {
      dist1 = tfDist;
      if (tfDist < 30)
      {
        t1 = millis();
        printf( "Time 1 : %04i s\n", t1);
      }
    }
    break;
  }
  while (sensor2serial.available() > 0)
  {
    if ( sensor2project.getData( tfDist)) // Get data from the device.
    {
      dist2 = tfDist;
      if (tfDist < 30)
      {
        t2 = millis();
        printf( "Time 2 : %04i s\n", t2);
      }
    }
    break;
  }
  T = t2 - t1;
  float Second = (float)T / 1000;
  printf( "Time difference : %0.3f s\r\n", Second);
  //printf( "Velocity : %04i s\n\n", velocity);
  //Serial.println(velocity);
  //printf( "Sensor 1 : %04i cm \r", dist1);   // display distance,
  //printf( "\r");                   // end-of-line.
  //printf( "Sensor 2 : %04i cm \r", dist2);   // display distance,
  //printf( "\r\n");                   // end-of-line.

  delay(50);

}