budryerson/TFMini-Plus #15: Stuck on checksum

safikka, 2020-07-06:
I am stuck in a checksum. Before that, I tried using the code for Arduino from the official Benewake and success to read the distance. Also i tried to read via GUI, and managed to read. But when i tried to use this library, its stuck on checksum. Any idea? Thanks

[image]

budryerson, 2020-07-06:
Hello Safikka.  Are you using the latest version of the TFMPlus library, v1.4.0?

I ran the library example code.  It did return a `TIMEOUT` error after the `OBTAIN_FIRMWARE_VERSION` command; but I believe that was because the device wasn't allowed enough time to execute the prior `SYSTEM_RESET` command.  So I added a `delay(300);` statement before the Get Firmware command and that fixed that.

Also, the ° (degree) symbol after temperature data did not display correctly.  That's always been trouble.  I should just get rid of it.

Everything else appeared to work correctly.

The last (checksum) byte of your data `59 59 E7 00 45 07 18 0A 06` should be `0x07` and not `0x06` so the error message is correct.

I don't know what to tell you, but I'm sure we can solve this.  Please give me more information.  If you will ZIP up your sketch and libraries and send them to me, maybe I can duplicate the error you're getting and we can figure this out together.

Sorry for the problems.

Bud Ryerson
bud at budryerson dot com

safikka, 2020-07-07:
Yes, i'm using the latest version 1.4.0
After trying to add delay() before get firmware, it's still stuck.
[image]
[image]

Here i attach the code and library
[TFMini Plus.zip](https://github.com/budryerson/TFMini-Plus/files/4882377/TFMini.Plus.zip)

Anyway, thanks for your help!

budryerson, 2020-07-07:
Safikka,

I see another error of mine.  On line 71 of the example sketch, please change `FRAME_250` to `FRAME_20` .
This only effects the displayed response when the command 'passes.'  It has no effect on data.

Did you see the 'hello' message: `TFMPlus Library Example - 18JUN2020`
And did the three sample commands (reset, firmware version, frame rate) respond correctly?
`System reset: passed`
`Firmware version: 2.0.3` (or whatever your version is)
`Data-Frame rate: 20`
If not, what were the results?

I notice you're using the software serial library.  I think our problem may be here.
Which Arduino are you using?

Bud

safikka, 2020-07-07:
Hello Bud,

Unfortunately i dont see anything from 'hello' message and three sample command.
The only result show in serial monitor is just checksum, so i guess it's stuck.
[image]

I'm using arduino pro mini and uno, and the result is still the same stuck.
Any idea?

budryerson, 2020-07-07:
I'm really sorry your having this problem.

The data looks messed up.  There should not be so many 0x59 characters.

Please restart the sketch with Serial Monitor Autoscroll 'off' so that we can see results from the begining of the sketch.

I understand that you are using both a Pro Mini and an Uno.  I have run this on a Pro Mini without problems.  Which pins are you using for SoftwareSerial on the Pro Mini?

You wrote "I tried using the code for Arduino from the official Benewake and success to read the distance."  If you would be so kind, will you please send me that 'official' code and your data?  You also wrote, "i tried to read via GUI, and managed to read." Can you send me your data from the GUI as well?

Please don't give up.  We can fix this.

Bud

safikka, 2020-07-07:
Hey Bud, thanks for the response!

Here result from the beginning
[image]
I don't know what "OTHER" stand for

if i restart the Arduino, here the result
[image]

Here the result in GUI
[image]

Here code from the official Benewake, I got this code from their github
[TFminiPlus-Arduino-master.zip](https://github.com/budryerson/TFMini-Plus/files/4886476/TFminiPlus-Arduino-master.zip)

I actually struggled to adjust the time of reading this tfmini, using delay() function in arduino interferes with sensor readings and make wrong measurement. So, i think using trigger mode and set the delay time it will be solved.
Seeing your library, I think it will be easier for me to set tfmini plus to trigger mode and set the reading time to a certain time because there is no guide to set it into trigger mode with Arduino.

budryerson, 2020-07-07:
`OTHER` stands for an error status other than one defined in the library.  On the serial terminal, it would normally display as `Status: OTHER`.

Your sketch `TFMP_tes.ino` does not have the `#include "printf.h"` statement in it.  And the `printf.h` file is not in your sketch folder.  Why?

safikka, 2020-07-08:
Hey Bud,

Thank you for your correction. Now, the library can display results. I don't know why the print.f library is missing in my code, maybe I forgot to add that library and delete it in the code. Here the result
[image]

Anyway thanks for your help! really appreciate it

closed 2020-07-08 by safikka