budryerson/TFMini-Plus #16: How to set reading time tfmini?

tuyulilang, 2020-07-07:
Hello Bud,
How do I set the tfmini reading time?
I want to read every 5 seconds. When using the delay function (5000), the reading results in lag and chaos.
I tried to set frame rate to 1Hz, then the tfmini read every 1 second. But, when I add delay function, it interrupts the sensor reading and makes the reading lag.
any suggestion?

budryerson, 2020-07-07:
I assume that you mean TFMini-Plus and not TFMini, which is a different device.

In UART (serial) communication mode, output from the device is continuous and free-running at the default or specified data frame rate.  According to the Benewake TFMini-Plus Product Manual, you can set the frame rate to `0` (zero) and then trigger the return of a single data frame by command.  But why do that?  The `getData()` function is designed to clear out the serial input buffer and then wait for the next good data frame to arrive.

I don't exactly understand your problems with "lag and chaos."  Can you show me your code and results?

Thanks,

Bud Ryerson
bud at budryerson dot com

modarocokdancok, 2020-07-10:
Hi bud!
After trying several experiments, it was able to use the command trigger. Thanks!
If we set the trigger mode say in 100 times in 1 second, will be the same as we set the frame rate 100Hz? which at 100hz frame rate mode, the reading becomes 100x sampling in one second.

budryerson, 2020-07-10:
I'm happy to hear that you were able to successfully use the TRIGGER_DETECTION command.  I think that the answer to your question is "Yes."

The internal chip is sampling at 4KHz.  The maximum recommended data frame return rate is 1KHz (4 samples per frame).  My understanding is that a lower frame rate, such as 100Hz, will result in a more accurate reading due to more samples for each frame.

I hope that helps.

closed 2020-10-24 by budryerson