budryerson/TFMini-Plus #18: Problemes upgrading to library 1.4.0

mprowe, 2020-07-25:
Hello again Bud,

I am compiling for the ESP32. Library 1.3.5  was working fine for me. However, I can not get 1.4.0 to compile?
This is how have I changed my sketch? 
1. Replaced
```
uint16_t tfDist     = 0;       // Distance measurement in centimeters (default)
uint16_t tfFlux     = 0;       // Luminous flux or intensity of return signal
uint16_t tfTemp     = 0;       // Temperature in degrees Centigrade (coded)
**with**
int16_t tfDist     = 0;       // Distance measurement in centimeters (default)
int16_t tfFlux     = 0;       // Luminous flux or intensity of return signal
int16_t tfTemp     = 0;       // Temperature in degrees Centigrade (coded)
```

2. Replaced all instances of
```
tfmP.printStatus(true); 
**with**
tfmP.printFrame(true);
```

3. Replaced the the 1.3.5 library with the 1.4.0 version using the Arduino IDE Library manager.

My errors:

```
Arduino: 1.8.12 (Linux), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

/******************* text truncated for brevity. Full log appended ********************/

Using library Wire at version 1.0.1 in folder: /home/mprowe/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/Wire 
Using library NHD_US2066 in folder: /home/mprowe/Arduino/libraries/NHD_US2066 (legacy)
Using library SPI at version 1.0 in folder: /home/mprowe/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/SPI 
Using library TFMPlus at version 1.4.0 in folder: /home/mprowe/Arduino/libraries/TFMPlus 
Using library WiFi at version 1.0 in folder: /home/mprowe/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi 
exit status 1
no matching function for call to 'TFMPlus::printFrame(bool)
```
[Error.log](https://github.com/budryerson/TFMini-Plus/files/4975482/Error.log)

Any suggestions will be appreciated.

Regards, Martin

budryerson, 2020-07-26:
Dear Martin,
First of all, download **v1.4.1** of the library.  There were a couple of bugs in v1.4.0.  I'm not sure they effected you, but...
Second, `printFrame()` does not take a parameter.  It's just `printFrame()` (no "true") for a `getData()` error and `printReply()` for `sendCommand()`.
Third,  delete the line `#include "printf.h"` from the example.  ESP32 includes `printf` in its standard IO package.  Also, **every** call to `printf()` needs to terminate with a "\n". For example: `printf("Hello!\n");`  "Hello!" will not display until a "\n" is sent.
After that, let's see where we are.  I have an ESP32 DEVKIT1 so please send me your code, if you can.
Tally ho!
Bud

mprowe, 2020-07-26:
Hi Bud,
Thank you very much for your prompt reply.

**v1.4.1** - Done.
**printFrame() does not take a parameter.** - Dho....! Silly me :(

Once again, many thanks for your help.

~Martin

closed 2020-07-27 by mprowe