budryerson/TFMini-Plus #19: TFMini (nonPlus) or TFLuna

ryancasler, 2020-08-12:
Will this library also work with the [TFMini ](https://www.amazon.com/Benewake-Single-Point-Ranging-Detectors-Rspberrypi/dp/B07BJZ18RC)or[ TF-Luna](https://www.amazon.com/Benewake-TF-Luna-Single-Point-Ranging-Interface/dp/B086MJQSLR)?  Thanks!!

budryerson, 2020-08-12:
The short answer is "No."

You might be able to use the TFMini-Plus library to receive serial data from the TF-Luna.  I have not tested it.  But **the TFMini-Plus-I2C library is definitely not compatible with the TF-Luna**.  The TF-Luna in I2C mode is entirely different.  I am working on libraries for the TF-Luna.

**The TFMini-Plus and TFMini-Plus-I2C libraries are not compatible with the TFMini**

ryancasler, 2020-08-12:
> I am working on libraries for the TF-Luna.

Wonderful!  Because the Luna matches up with my requirements for a lot of reasons (sensing distance and cost) so I would much rather use that but I'm not quite skilled enough to write the Arduino code from scratch.  I'll keep an eye out for it.  Thanks!!

closed 2020-08-12 by ryancasler