budryerson/TFMini-Plus #20: weird return values

ZorchDiabetes, 2020-10-01:
Hi Bud!

Love your library.  I'm having what I hope is a small problem getting the TFMini-Plus to talk to a Madunio Lora Radio on SoftwareSerial on pins (D4,D5)

It's doing this:

TFMPlus Library Example - 18JUN2020
System reset: Status: TIMEOUT
 F6 04 60 09 FF 00 00 00
Firmware version: Status: TIMEOUT
 A3 00 FA 04 B0 41 F8 00
Data-Frame rate: Status: TIMEOUT
 00 F5 04 60 09 BF 00 00
Status: HEADER
Data: FF 58 99 69 80 08 C0 42 DF
Status: HEADER
Data: 58 B9 A3 80 1F 08 4B C4 FF
Status: HEADER
Data: 58 99 A3 80 1F 08 53 C4 FF
Status: HEADER
Data: 58 99 A3 00 F4 04 60 09 EF
Status: HEADER
Data: 58 99 A3 00 F5 04 60 09 BF
Status: HEADER
Data: 58 99 A3 00 F6 04 60 09 FF
Status: HEADER
Data: 58 99 B3 00 F8 04 B8 41 F0
Status: HEADER

budryerson, 2020-10-02:
It looks like a serial problem, Zorch.  Are you using only one Software Serial port?  Are there any pin limitations on the Maduino (change interrupt)?   Why D4 and D5?  Please send me your code?  Thanks,  bud at budryerson dot com

ZorchDiabetes, 2020-10-02:
Bud I think you're right:  I'm using a 3.3V 8MHz Maduino Pro Mini and from what I've read, they can't do 115200 using SoftwareSerial.  I need to find a way to lower the baud or change to I2C.  Any suggestions which will be lower burden?  This is for monitoring water level in our well, which ran dry 10 days ago after a very dry summer here in Nova Scotia.

Regards,

Lane

budryerson, 2020-10-02:
I don't have a problem with Software Serial at 115200.  SS does have a problem with RX on certain pins on certain processors.  I do recommend I2C.  Try to keep your wiring short and don't use ribbon cable (conductors side-by-side).
  
  
   
  
  
   Did you see the 'readme' on the github page?
   
  
  
   https://github.com/budryerson/TFMini-Plus
  
  
   
  
  
   Keep me posted.
   
  
  
   
  
  
   Bud

ZorchDiabetes, 2020-10-13:
Bud I wanted to let you know that I got everything working, thank you very much.  I used a Mega to switch to I2C, which then worked just fine on the Maduino LORA radio / pro mini.  I'm now [reading the well water level](https://thingspeak.com/apps/matlab_visualizations/369966) no problem, although I'm not 100% sure that I've got it polling properly.  Is there any example code for that?  Power consumption is an issue, with little sun in Nova Scotia this time of year...
[image]
[image]

Regards,

Lane

budryerson, 2020-12-31:
That's truly clever.  I'm happy to hear that it's working.
A larger solar panel is always helpful, especially on cloudy days. Did you make it past the solstice?
Best regards,
Bud Ryerson
San Francisco

closed 2020-12-31 by budryerson

ZorchDiabetes, 2021-01-01:
Hi Bud,

Thanks for getting back to me.  I was able to get the TFmini working again
- when I bricked it with I2C, it reverted to serial (factory reset)?  I
used a Mega to get it back into I2C mode before reconnecting it to the
LORA, which only has one serial port.
https://www.makerfabs.com/maduino-lora-radio-433m-868m-915m.html

I brought the system back home to SoCal with me in November because my
father decided to drill a new well, obviating the need for monitoring the
old well :-).

The project was quite fun; I learned a lot.

Warm regards,

Lane