budryerson/TFMini-Plus #21: Not working on ESP8266 chip

octimot, 2020-10-23:
I tried to run the code on a Wemos D1 Mini board.

First, the compiler threw an error, which I fixed like this:

Line 30 in print.h:
fdevopen( &serial_putc, 0 );

Needs to be changed to:
Serial.setDebugOutput(true);

This seems to be the proper way to redirect stdout on ESP8266 chips - a conditional based on the chip type might be useful in the code in the future.

But even after uploading the revised code to the board, I can't get it to communicate with the TFMini Plus (which works well on an Arduino Uno).

Did somebody have any luck with this?

octimot, 2020-10-23:
I dug a bit deeper and it could be that SoftwareSerial triggers a crash on ESP8266 chips. 

On https://github.com/opensensinglab/tfmini/issues/6 they recommend using the ESPSoftwareSerial for the TFMini.

Not sure how I could implement that info for testing. Help? 🤓

octimot, 2020-10-24:
I managed to make it work, by using the main Serial class for communicating with the TFMini Plus.

Basically:
- I disabled any sort of Serial output (entire TFMPlus::printFrame() and TFMPlus::printReply() ),
- commented out all printf functions,
- and removed the inclusion of SoftwareSerial since it's not needed anymore.

I want to output tfDist to an OLED shield, so, for my purpose, I can still see the measured results even without the serial connection to the computer.

But if someone wants to tackle this, it would be nice to have it fixed. If you need the altered code, just let me know.

budryerson, 2020-10-24:
I was not aware of the Wemos D1 Mini board.  Now I am.  Thank you.

I like `printf()` because it enables compact, easy to read code; but I didn't create it and, as you noted, problems can arise.
`printf()` may be already included in the Arduino IDE for ESP8266 chips, in which case you would not need to call _printf.h_.  ESP8266 `printf()` may behave differently in that a `\n` needs to be sent before any output occurs.  I agree that it's sometimes more reliable to replace every `printf()` with `Serial.print()`.

I also agree that SoftwareSerial brings it's own set of problems and that's why I do not understand why you want to communicate with the TFMini-Plus in the UART mode.  Why not use I2C mode?

I would love to see your code.  Thanks,

Bud Ryerson
San Francisco

closed 2020-12-31 by budryerson