budryerson/TFMini-Plus #22: Change to I2C returning Serial

runfastman, 2020-12-22:
I make a call in the beginning of the code to change to I2C but it always returns "Serial" but no error with the SetCommand. Do I need a parameter with the SET_I2C_MODE command?

**Using Example code** 

void setup()
{
    Serial.begin(115200);   // Intialize terminal serial port
    delay(20);               // Give port time to initalize
    printf_begin();          // Initialize printf.
    printf("\r\nTFMPlus Library Example - 18JUN2020\r\n");  // say 'hello'

    mySerial.begin(115200);  // Initialize TFMPLus device serial port.
    delay(20);               // Give port time to initalize
    tfmP.begin( &mySerial);   // Initialize device library object and...
                             // pass device serial port to the object.

    // - - - - - - - - - - - - - - - - - - - - - - - -
    printf( "SET_Mode: ");//SET_I2C_MODE, SET_SERIAL_MODE
    if( tfmP.sendCommand(SET_I2C_MODE , 0))
    {
        tfmP.printReply();
    }
    else 
    {
      printf( "Failed: ");
      tfmP.printReply();
    }
//.....

**Output**

SET_Mode: Status: SERIAL
 00 00 00 00 00 00 00 00

budryerson, 2020-12-31:
I'm sorry for the delay.  You've probably figured it out by this time.
But will you please send me all your code so I can duplicate your problem?
Thanks,
Bud

runfastman, 2021-01-01:
Here is the code.

"sendCommand" for I2C returns "Ready" but the readings continue via RX

[image: image.png]

budryerson, 2021-01-01:
There was no attachment with your last message.
Please send me your entire sketch (_"all your code"_) so that I can duplicate your problem.
You may send the archive to my email `bud@budryerson.com`
Please include images of your setup and cabling, if you think that will be helpful.
Thanks,
Bud

budryerson, 2021-01-02:
1)  You listed only two lines of output:  
`SET_Mode: Status: SERIAL`
`00 00 00 00 00 00 00 00`
Is that it?  Only two lines?  No "Hello"?
Something like:
`TFMPlus Library Example - 18JUN2020`

2)  Your Software Serial assignment of Pins 0 & 1
` SoftwareSerial mySerial( 0, 1);`
appears to use the same pins as the hardware serial port.
Are you sure about those pin assignments?
Which Arduino board (and revision) are you using?

3)  Is the Arduino working? Is the sensor working?
Is the Arduino working okay with the sensor in Serial mode?

We'll fix this.

Bud

runfastman, 2021-01-02:
1. I am confused. There are several lines of output. I used your example
and added this at line 50
// - - - - - - - - - - - - - - - - - - - - - - - -
    printf( "SET_Mode I2C: ");
    if( tfmP.sendCommand(SET_I2C_MODE , 0))//SET_I2C_MODE, SET_SERIAL_MODE
    {
        tfmP.printReply();
    }
    else
    {
      printf( "Failed: ");
      tfmP.printReply();
    }

//This is the output I get
[image: image.png]

2. I was following the instructions "Delete the comment slashes // on lines
37 and 38 to invoke the library, and be sure to choose // the correct RX
and TX pins: pins 10 and 11"
I am using - "Nano V3.0 Module ATmega328P 5V 16MHz CH340G" with D0 & D1 as
TX & RX

3. Yes, the sensor is working and giving me the correct distance with
serial, but I prefer to use I2C to simplify multiple sensors from the same
board.

budryerson, 2021-01-02:
1.  I apologize for any confusion.  Your [images] are not showing up here.  Please save your output into a text file and attach or send me the file.

2.  The D0 & D1 pins on the **Nano V3.0 Module ATmega328P 5V 16MHz CH340G** are the hardware serial lines that you use to program and communicate with the **Nano** from your IDE (running on your PC).  The **Nano** has an onboard USB to TTL converter (CH340G) that requires the use of these pins.  You cannot use these same pins to communicate with your **TFMini-Plus**.  That's why we use the Software Serial library - because the ATmega328P does not have a second hardware serial port.  I believe that pins 9 & 10 will work for that purpose (11 & 12 are for the SPI interface) ; but please check the Software Serial documentation for your particular board.
    
      You say that your **Nano** and your **TFMini-Plus** are working together just fine in Serial Mode.  I don't see how that is possible, so I must really be misunderstanding something.  Again, please send me a text file of your output from running the example sketch in the Serial Mode.  Pictures of your wiring would be helpful, too.

3.  You are correct that the I2C Mode is the way to go.  It will allow you to use more than one **TFMini-Plus** on the same bus.

Bud Ryerson
San Francisco

runfastman, 2021-01-02:
Thanks so much for your help with this!

1. Here is the output and a wire pic. The Distance changes if I put my hand
above it. I do have to unplug the tx/rx wires to flash but then I plug back
in and hit the rest button and monitor the output.
TFMPlus Library Example - 18JUN2020
SET_Mode I2C: Status: READY
 00 00 00 00 00 00 00 00
System reset: Status: TIMEOUT
 44 0A 90 09 6F 00 00 00
Firmware version: Status: TIMEOUT
 D6 00 38 0A 90 09 C3 00
Data-Frame rate: Status: TIMEOUT
 00 37 0B 90 09 62 00 00
Dist:0214cm Flux:02616 Temp:50°C
Dist:0214cm Flux:02611 Temp:50°C
Dist:0214cm Flux:02615 Temp:50°C
[image: MiniWiring.jpg]

2. When I change to "SoftwareSerial mySerial(9,10);" Pins D6 D7, the result
from 12c sendCommand is "Serial" instead of "Ready" and I don't get any
distance readings.

TFMPlus Library Example - 18JUN2020
SET_Mode I2C: Status: SERIAL
 00 00 00 00 00 00 00 00
System reset: Status: TIMEOUT
 00 00 00 00 00 00 00 00
Firmware version: Status: TIMEOUT
 00 00 00 00 00 00 00 00
Data-Frame rate: Status: TIMEOUT
 00 00 00 00 00 00 00 00

[image: MiniWiring9_10.jpeg]

budryerson, 2021-01-02:
Your images are still not showing.

You wrote that, _"the sensor is working and giving me the correct distance with serial."_  So I asked to see the output from `TFMP_example.ino`, the Serial Mode example sketch, running correctly.  I don't see that.  In the output you show above, every command has failed.

You wrote, _"I do have to unplug the tx/rx wires to flash but then I plug back in and hit the rest button and monitor the output."_
Why are you plugging and unplugging wires?  Which _"tx/rx wires"_ are you talking about?  How do you _"monitor the the output?"_

I get the uneasy feeling that you are still confusing hardware and software serial ports.  The hardware serial port is for programming and monitoring ONLY.  The Software Serial port is for communicating with the Lidar.  Am I wrong?

Please let me know.

Bud

runfastman, 2021-01-03:
1. Correct , the other commands seem to timeout, but the distance is
working.

2. *Why are you plugging and unplugging wires? Which "tx/rx wires" are you
talking about?*
*The white to tx and the green from RX (I shared pics on google drive with
you)*

3. *How do you "monitor the the output?" *
*I use the Serial monitor from arduino UI.*

4. *I get the uneasy feeling that you are still confusing hardware and
software serial ports. The hardware serial port is for programming and
monitoring ONLY. The Software Serial port is for communicating with the
Lidar. Am I wrong?*
*I may be, I have only used the RX and TX for a GPS chip. before.  I tried
using D6 and D7 (See other pic on Google drive) and I don't get any output
the serial monitor. I have an Uno, but it also only has the one serial line
on 0 & 1. *

budryerson, 2021-01-03:
Please understand that the Lidar device communicates with the Arduino - and the Arduino then turns around and communicates with the user (you).  This requires two independent data channels.  The USB cable connects the Arduino with your computer.  It uses the D0 & D1 (TX/RX) pins.  You cannot **EVER** use those to connect to the Lidar.  You must use the Software Serial pins.  ***BUT***  As soon as you send a `SET_I2C_MODE` command to the Lidar, that device can no longer communicate in serial mode.  So **DO NOT** send any commands to change the communication mode until after we've got everything working in serial mode.  Okay?
I am still not getting your files or images.  Please contact me further by email at:
bud-at-budryerson-dot-com

closed 2021-03-17 by budryerson