budryerson/TFMini-Plus #25: TF Mini Plus Raspberry Pi library

mvstimpson, 2021-02-01:
Any chance you have a library for Raspberry Pi for the TF Mini Plus? Getting checksum errors and not sure if there's a code problem or a sensor problem.

budryerson, 2021-02-03:
Sorry, no, I do not yet have a Raspberry Pi library.  Someday soon, though.
How are getting checksum errors?

budryerson, 2021-03-24:
Please take a look at the Python module, **TFMini-Plus_python**.
This version is for serial mode only.  It's not distributed to PyPI yet.
I'd appreciate your feedback.

budryerson, 2021-06-09:
There are two Python TFMini-Plus libraries now on PyPI:  `tfmplus` and `tfmpi2c`.

For example, in the command line:
```
python -m pip install tfmplus
```

And in your python script, something like this:
```
import tfmplus as tfmP   # Import the 'tfmplus' module and
from tfmplus import *    # the command and parameter definitions
```
Let me know how that works for you.

Bud Ryerson
San Francisco

mvstimpson, 2021-06-09:
Thank you for supporting this! As of right now we have moved on to the TF03
instead of the TFMini Plus. It's much more expensive, but we need the
longer distance, plus the 03 works in rain and snow.

Michael Stimpson

Principal Partner – Triple Talon, LLC

***@***.*** ***@***.***>

www.tripletalon.com

480.208.3658

closed 2021-12-01 by budryerson