budryerson/TFMini-Plus #26: 'fdevopen' was not declared in this scope

jimbobhunter, 2021-02-13:
Hi. Looking forward to using the library but I get this error when trying to compile for M5Atom.

budryerson, 2021-02-20:
Jim Bob,
I look forward to helping, but I'm going to need a little more information.
Please zip up the sketch that your using and a file with the output that your getting.
Email it to me:  bud-at-budryerson-dot-com.
We will fix this.
Thanks,
Bud Ryerson
San Francisco

jimbobhunter, 2021-02-21:
> Jim Bob,
> I look forward to helping, but I'm going to need a little more information.
> Please zip up the sketch that your using and a file with the output that your getting.
> Email it to me: bud-at-budryerson-dot-com.
> We will fix this.
> Thanks,
> Bud Ryerson
> San Francisco

Sent, thanks Bud!

closed 2021-03-17 by budryerson

default-student, 2023-09-28:
I dont know how this fixes the problem.
Im afraid i ran into it as well

budryerson, 2023-09-28:
Back in February of 2021, Jim Bob Hunter sent me his error messages and his code as I requested.  I sent him the following suggestion.
```
James,

You are not using 'printf' but you left in all the odd formatting characters.
Such as in Line 74:
    Serial.print( "%1u.", tfmP.version[ 0]);
"%1u." is an instruction to print the data as a single character of an unsigned integer.
It doesn't make any sense to your compiler.

You should format the data into a string and then print the string.

Bud 
```
Maybe yhou have the same problem and maybe the same advice will help you as well.  Please  let me know.

Bud Ryerson
San Francisco

default-student, 2023-10-02:
Replacing all the printf's with Serial.print fixed it