budryerson/TFMini-Plus #27: TFmini plus timeout?

didi1150, 2021-03-06:
I'm getting the messages: 

TFMPlus Library Example - 18JUN2020
System reset: Status: TIMEOUT
 00 00 00 00 00 00 00 00
Firmware version: Status: TIMEOUT
 00 00 00 00 00 00 00 00
Data-Frame rate: Status: TIMEOUT
 00 00 00 00 00 00 00 00
Status: HEADER
Data: 00 00 00 00 00 00 00 00 00

And then it continues printing the last two lines.

I'm using the example code.

budryerson, 2021-03-06:
It appears that you are not communicating with the device.
Forgive me, but please answer a couple of necessary questions:
1) Is the TFMini-Plus turned on?  Can you see the red glow?
2) Did the device communicate correctly with the Benewake GUI?
3) Which Arduino are you using?
4) Are you using hardware or software serial?
5) Are the RX./TX lines are connected correctly?
6) Can you send me a picture of your setup?
I'm sure we'll fix this.
Bud Ryerson
San Francisco

didi1150, 2021-03-07:
Im using the arduino mega. And is it possible to change the pins of the tfmini plus? Because you could set them in the ide when using the tfmini lib.

didi1150, 2021-03-07:
Oh and the tfmini plus doesnt glow

didi1150, 2021-03-07:
Oh and Im using software serial. As far as i know the rx tx lines are connected correctly.

didi1150, 2021-03-07:
Oh and also I modified the Serial2 to mySerial, because it always says 'Serial2' was not declared in this scope when I use Serial2.

didi1150, 2021-03-07:
Ok Ive done it, the sensor is onw working

didi1150, 2021-03-07:
I just connected rx and tx wrongly.

closed 2021-03-07 by didi1150