budryerson/TFMini-Plus #29: printf.h: No such file or directory

jax200, 2021-03-17:
Hello

I am relatively new to this.   I have connected up a TFMini Plus to an Arduino UNO, added this library to /Arduino/libraries, then attempted to run TMFM_example.ino.   On verify I got the following, so not sure what to do.   Thanks!

`TFMP_example:25:10: fatal error: printf.h: No such file or directory
 #include "printf.h"   // Modified to support Intel based Arduino
          ^~~~~~~~~~
compilation terminated.
exit status 1
printf.h: No such file or directory
`

budryerson, 2021-03-17:
Make sure that the file `printf.h` is in the same folder with your sketch.
`printf.h` is included with the example sketch.

Surfer2010, 2021-03-24:
In file included from C:\Users\Dom\Documents\Arduino\TFMini-Plus_Budryerson\example\TFMP_example\TFMP_example.ino:25:0:
sketch\printf.h: In function 'void printf_begin()':
sketch\printf.h:30:29: error: 'fdevopen' was not declared in this scope
   fdevopen( &serial_putc, 0 );
                             ^
exit status 1
Fehler beim Kompilieren für das Board DOIT ESP32 DEVKIT V1.

ripan-uppsala, 2021-04-17:
I am succesfully using your code, thanks :), but instead of printf.h used the official download "PrintEx.h" in this case on a Nano Every.

#include "PrintEx.h"                           // To support printf
StreamEx mySerial = Serial;               // Creat a StreamEx object

and renamed all printf instances to mySerial.printf

closed 2022-04-04 by budryerson