budryerson/TFMini-Plus #30: How to use TFmini plus distance reading as a threshold?

CrowX35, 2021-03-23:
Hi! I'm working on an obstacle avoidance system using TFmini plus. I want to know how do you obtain the getDistance value as an integer in order to turn on and off an LED depending upon the distance. My code doesn't seem to work.

[image]

budryerson, 2021-03-24:
Dear Crow,

The `getData()` function returns a PASS/FAIL boolean value.  The variable `tfDist` is seen by `getData()` as a pointer.  The function uses the pointer to set the value of the variable.  Your code could look more like:
```
tfmp.getData( tfDist)  # ignore the boolean return
val = int( tfDist)     # recast the variable as an integer
```
or skip a step and just send `val`:
```
uint16_t val
tfmp.getData( val)
```
Please let me know whether that helps or not.

Bud Ryerson
San Francisco

CrowX35, 2021-03-24:
It works. Thank you so much!

closed 2021-03-24 by CrowX35