budryerson/TFMini-Plus #31: Failure rate

ttttttom, 2021-03-23:
Hi budryerson,
I am using your library for 1 TFMini Plus with SoftwareSerial and an Arduino Uno. I was wondering why so many readings are bad. So I wrote a little code to show it. I give you some random lines of that output:

Test TFMini Plus (203) at 500Hz 500x -> Failure rate: ss 30.20
Test TFMini Plus (203) at 1000Hz 500x -> Failure rate: sf 29.00
Test TFMini Plus (203) at 20Hz 100x -> Failure rate:  29.00
Test TFMini Plus (203) at 100Hz 500x -> Failure rate: ss 32.20
Test TFMini Plus (203) at 200Hz 500x -> Failure rate: ss 33.80
Test TFMini Plus (203) at 500Hz 500x -> Failure rate: ss 37.20
Test TFMini Plus (203) at 1000Hz 500x -> Failure rate: s 30.20
Test TFMini Plus (203) at 20Hz 100x -> Failure rate: sf 32.00
Test TFMini Plus (203) at 100Hz 500x -> Failure rate: ss 33.00
Test TFMini Plus (203) at 200Hz 500x -> Failure rate: ss 24.80
Test TFMini Plus (203) at 500Hz 500x -> Failure rate: ss 14.80
Test TFMini Plus (203) at 1000Hz 500x -> Failure rate: s 20.80
Test TFMini Plus (203) at 20Hz 100x -> Failure rate:  24.00
Test TFMini Plus (203) at 100Hz 500x -> Failure rate: sssss 22.60
Test TFMini Plus (203) at 200Hz 500x -> Failure rate: ss 21.40
Test TFMini Plus (203) at 500Hz 500x -> Failure rate: sss 21.00
Test TFMini Plus (203) at 1000Hz 500x -> Failure rate: s 22.60
Test TFMini Plus (203) at 20Hz 100x -> Failure rate:  21.00

s=system reset failed once
f=set frame rate failed once
Failure rate= failed readings/total readings

The distance is just 2,2 m to a white wall. And it is indoors.
What do you think? Would that be normal for a TFMini Plus?

Thanks in advance!
Tom

PS: If it is of any help, I could give that test code to you.

budryerson, 2021-03-24:
Tom,
That seems excessive.
Do you get the same results with a hardware serial port?  Have you tried I2C mode?
Sure, I'd love to see your code... and maybe pictures of your setup, too, if you can.
Bud Ryerson
San Francisco

ttttttom, 2021-03-26:
Dear Bud, 
thank you for your answer. I did not test I2C or HW-serial yet, but I will do. Until then I give what I have:
[image]
Notice: The failure rate does not change when using a 5V power supply (green cable).

[tfminiplus.ino.zip](https://github.com/budryerson/TFMini-Plus/files/6212627/tfminiplus.ino.zip)

Tom
Aachen, Germany

budryerson, 2021-03-28:
Tom,
1) You do not match the lidar device's frame-rate with the sample rate of your sketch.  If the device frame-rate is set to 20Hz then your sketch should not sample any faster than that.  In my example sketch, I add a 50ms delay to the main loop. In your sketch, an delay appropriate to the frame-rate should be included in your sampling function `test_run()`.
2) The `while()` loop in 'tfminiplus_reset() is banging away at the device much too quickly.  The lidar needs several hundred millisecconds to reset.  A 500ms delay should be enough.  And you only have to do it once.  If everything is working, it should never fail.
3) Likewise with `tfminiplus_setframerate()`.  That command does not need as much time as a reset command; but try increasing the delay.  Failures should be zero.  Also, for good housekeeping, you might add a SAVE_SETTINGS command (plus delay) to follow the frame rate change.
4) In the picture you included (thank you!) your lidar lens seems very smudged.  It is unlikely that is a problem, but I'd clean it anyway.
5) You might want to take advantage of the library error codes or `status` byte.  Sometimes it can be informative to see how the command is failing and what the actual serial data return looks like.
Good luck.  Let me know what happens and if you have any questions.
Bud Ryerson
San Francisco

ttttttom, 2021-04-10:
Thank you for your precise answer, Bud. 
Today, I checked your suggestions and here are my answers:

(4.) First I cleaned the sensor. No other change. Failure rate is still the same.

(1.) I included the suggested delay. But there is no significant change in the failure rate. Here some results:
Test TFMini Plus (203) at 500Hz 500x (pause=10ms)-> Failure rate: s 29.60
Test TFMini Plus (203) at 500Hz 500x (pause=0ms)-> Failure rate: s 28.40
Test TFMini Plus (203) at 1000Hz 500x (pause=10ms)-> Failure rate: s 31.00
Test TFMini Plus (203) at 1000Hz 500x (pause=0ms)-> Failure rate:  29.00
Test TFMini Plus (203) at 20Hz 100x (pause=50ms)-> Failure rate:  26.00
Test TFMini Plus (203) at 20Hz 100x (pause=0ms)-> Failure rate: ss 28.00
Test TFMini Plus (203) at 100Hz 500x (pause=10ms)-> Failure rate: ss 22.20
Test TFMini Plus (203) at 100Hz 500x (pause=0ms)-> Failure rate: ss 28.80
Test TFMini Plus (203) at 200Hz 500x (pause=10ms)-> Failure rate: ss 22.20
Test TFMini Plus (203) at 200Hz 500x (pause=0ms)-> Failure rate: s 28.40
Test TFMini Plus (203) at 500Hz 500x (pause=10ms)-> Failure rate: sssss 27.00
The explanation: The sensor reading takes the sample time already. So the focus comes back after the reading. An additional delay is therefore not necessary.

(2.) I added a 500ms delay. As this code is for testing, a repeated reset should be okay and not lead to misbehavior. Of course, when using the device for a project, I would not do it. Here the output:
Test TFMini Plus (203) at 500Hz 500x (pause=0ms)-> Failure rate: ss 28.40
Test TFMini Plus (203) at 1000Hz 500x (pause=10ms)-> Failure rate: s 28.80
Test TFMini Plus (203) at 1000Hz 500x (pause=0ms)-> Failure rate: f 27.40
Test TFMini Plus (203) at 20Hz 100x (pause=50ms)-> Failure rate:  25.00
Test TFMini Plus (203) at 20Hz 100x (pause=0ms)-> Failure rate: sf 42.00
Test TFMini Plus (203) at 100Hz 500x (pause=10ms)-> Failure rate: s 25.60
Test TFMini Plus (203) at 100Hz 500x (pause=0ms)-> Failure rate: s 27.80
Test TFMini Plus (203) at 200Hz 500x (pause=10ms)-> Failure rate: sfff 23.40
Test TFMini Plus (203) at 200Hz 500x (pause=0ms)-> Failure rate: s 30.60
Test TFMini Plus (203) at 500Hz 500x (pause=10ms)-> Failure rate: sf 30.20
Test TFMini Plus (203) at 500Hz 500x (pause=0ms)-> Failure rate: sf 29.40
Test TFMini Plus (203) at 1000Hz 500x (pause=10ms)-> Failure rate: s 25.00

(3.) I added 500ms -> There are still failures when changing. I added `this->sendCommand(SAVE_SETTINGS, 0);` after successful `sendCommand(SET_FRAME_RATE, freq)` -> also no change in failure behavior.

(5.) I checked the status. It is always a CHECKSUM error.

I will continue testing using hardware serial. I still hope it is not the quality of the device. Maybe you could test one of your sensors with the test-class to clarify that point??
Best regards
Tom
PS: 
[tfminiplustest.zip](https://github.com/budryerson/TFMini-Plus/files/6290160/tfminiplustest.zip)

ttttttom, 2021-04-11:
Ok, this is interesting. 
1) Using hardware serial, there are no failures at all. 
2) An additional delay in the loop for sampling is not necessary, as I wrote above. 
3) You can reset the device as often as you like. A delay of 500ms after reset command is necessary as you said above. 

Update:
[tfminiplustest.zip](https://github.com/budryerson/TFMini-Plus/files/6291690/tfminiplustest.zip)

Regards and thanks again!
Tom

closed 2021-04-26 by ttttttom