budryerson/TFMini-Plus #34: sendCommand() not working with Arduino MKR GSM 1400

ibrcic, 2021-05-26:
Hello,

I am using the library with TFMini Plus and Arduino MKR GSM 1400 board.
I connected the sensor to TX and SX serial ports on my board. I can get the distance data without a problem as that is working as expected.

What I do have problem with is sending ANY command to the sensor itself. The execution just halts and nothing is returned by `sendCommand()` function. Here is an example code that does not work (I want to check firmware version, but ultimately want to change to I2C as I'll connect multiple sensors).

Here is the code example (I'm using Serial.print() as printf is not working)

```
#include <TFMPlus.h>
#include "printf.h"

TFMPlus tfmP;
void setup()
{

  Serial.begin(9600);
  while (!Serial);
  delay(20);

  Serial1.begin(115200);
  delay(20);               // Give port time to initalize
  tfmP.begin( &Serial1);
  
  delay(500);

  if (tfmP.sendCommand(OBTAIN_FIRMWARE_VERSION, 0))
  {
    Serial.println("Printing firmware version:");
  }
  else
  {
    Serial.println("Error printing version");
    tfmP.printReply();
  }
}

void loop()
{

}
```

When I run this, nothing is printed to the serial monitor, the execution apparently stops in `tfmP.sendCommand(OBTAIN_FIRMWARE_VERSION, 0)` part of the code.

budryerson, 2021-05-27:
Hello,

Thank you for the sketch.  With no output at all, the execution must stop somewhere before the first `Serial.println()` instruction. Could you add a `Serial.println("Hi there!");` after `Serial.begin()` and another after `Serial1.begin()`?

Although I am not familiar with the Arduino MKR GSM 1400, I understand that MKR boards address pins D13 and D14 as Serial1.
Would you be kind enough to send me a picture of your setup?

Thanks,

Bud Ryerson
San Francisco

budryerson, 2021-05-27:
Great!  Thanks.
I tried to make this library as bullet-proof as I could imagine - so that it would return something and not just hang.  I'm really sorry about the situation. I want to help, and I want to understand what's going on.

I am mystified.  You got past `Serial1.begin();` but that doesn't return much information.  Now please show me how you get distance data.  Maybe after `delay(500);` you could add some code like this?
```
int16_t tfDist = 0;
tfmP.getData( tfDist);
Serial.println( tfDist);
```
And if that doesn't work, try something more basic, like:
```
if( Serial1.available()) Serial.println( Serial1.read());
else Serial.println( "Serial1 data not available.");
```
I hope did that right. I'm not at the lab.  Please Let me know what happens.

Thanks,
Bud

ibrcic, 2021-05-28:
Hi,

I managed to resolve my issue, although not directly. First I used the provided GUI to change the communication mode to I2C (because that's what I'll be using for this project). After changing the mode, sending commands still didn't work. However, when I powered my Arduino from bench power supply in addition to connecting it to the PC via USB, everything started behaving normally.

I don't know if power was the culprit with serial communication also, but currently everything is working as expected with I2C version of this library, so this issue can be closed.

Thank you for your help!