budryerson/TFMini-Plus #35: Signal saturation Data - Error

Surfer2010, 2021-05-28:
Hello,
i'm using this library in arduino

TFMPlus I2C Library 1.5.0
11:56:15.300 -> Firmware version: 2.0.3
11:56:15.814 -> Data-Frame rate: 20Hz.
11:56:17.965 ->  Status: Signal saturation Data: 59 59 00 00 FF FF E8 08 A0

what can cause this problem? I checked the cables and all seems to be fine
Not sure what this error/status message means?!

caused by this code: (I2C Mode)
tfmP.status == TFMP_READY

budryerson, 2021-05-28:
Hey Surfer,
It's supposed to indicate an environmental problem, such as the Lidar looking into a mirror or the sun.
What's your setup?  Is your power okay?  Which Arduino are you using?  Which Lidar?
Will you please send me see **all** your code, a text file of your output and maybe a photo?
Thanks,
Bud Ryerson
San Francisco

Surfer2010, 2021-05-30:
Hello Bud,
thanks to your answer. The sensor is built into a tube to measure a water level. Since it won't detect water, there is a styrophom floating on top of the water.
All this is inside a concrete tube. Orange tube is the tube the sensor is on top.
[image]
[image]
[image]
[image]

[image]

It has been working. And this error MIGHT occour every time the pump is emptying the water. But this should only take about 10-15 min but the sensor is 45-60min "offline" Showing those error codes.

17:16:12.488 -> TFMPlus I2C Library 1.5.0
17:16:13.474 -> Firmware version: 2.0.3
17:16:13.989 -> Data-Frame rate: 20Hz.
Status: Signal saturation Data: 59 59 00 00 FF FF 00 09 B9

Not sure if it is EM-interferrance or something else.

definatly no mirror, nor any daylight.

Thanks for your help.

`void setup_TFminiPlus() {
  Wire.begin();            // Initialize two-wire interface
  printf( "\r\nTFMPlus I2C Library 1.5.0\r\n");  // say 'hello'

  // Send some example commands to the TFMini-Plus
  // - - Perform a system reset - - - - - - - - - - -
  printf( "System reset: ");
  if( tfmP.sendCommand( SOFT_RESET, 0))
  {
      printf( "passed.\r\n");
  }
  else tfmP.printReply();
  // - - Display the firmware version - - - - - - - - -
  printf( "Firmware version: ");
  if( tfmP.sendCommand( GET_FIRMWARE_VERSION, 0))
  {
      printf( "%1u.", tfmP.version[ 0]); // print three single numbers
      printf( "%1u.", tfmP.version[ 1]); // each separated by a dot
      printf( "%1u\r\n", tfmP.version[ 2]);
  }
  else tfmP.printReply();
  // - - Set the data frame-rate to 250 - - - - - - - - -
  printf( "Data-Frame rate: ");
  if( tfmP.sendCommand( SET_FRAME_RATE, FRAME_20))
  {
      printf( "%2uHz.\r\n", FRAME_20);
  }
  else tfmP.printReply();
  // - - - - - - - - - - - - - - - - - - - - - - - -
  delay(500);            // And wait for half a second.
}

void readTFminiPlus(){
  tfmP.getData( tfDist, tfFlux, tfTemp); // Get a frame of data
    if( tfmP.status == TFMP_READY)         // If no error...
    {
        Serial.println(" \t");printf( "Dist:%04icm ", tfDist);   // display distance,
        Serial.print(" \t");Serial.print(" \t");printf( "Flux:%05i ",   tfFlux);   // display signal strength/quality,
        Serial.print(" \t");Serial.print(" \t");printf( "Temp:%2i%s",  tfTemp, "°C" );   // display temperature,
        printf( "\r\n");                   // end-of-line.
    }
    else tfmP.printFrame(); // Otherwise, display error and data frame
    
    delay(50);    //  Run loop at approximately 20Hz.
}`

budryerson, 2021-05-30:
[surfer.ino.txt](https://github.com/budryerson/TFMini-Plus/files/6566846/surfer.ino.txt)

Dear Surfer,

Attached is a file I made from the snippet of code you included above.  Please send me **ALL** of your code.  Send me the files themselves (not excerpts), the version of the libraries you are using, **AND** a text file of your output.  Wrap them all up into an archive and send that to me, please.

The pictures are great.  Your application is unusual.  The TFMini-Plus would not be my first choice for this job.  The beam from the Lidar spreads out with distance.  At 2 meters, the "light spot" is 12cm (5 inches) wide and I have to seriously wonder how the light behaves inside an orange tube.  Also, I can see specular reflections around the styrofoam, presumably from the water.  That's not good.  I would also worry about any water on the surface of the styrofoam float.

Have you tried any other sensor for this application?  I think that something like the Sharp GP2Y0A line of products might be better suited, or even the HC-SR04 ultrasonic transducer.

You are using I2C mode.  Have you considered the length of your signal lines?  There is some controversy, but it's safe to say they should be as short as possible.  Have you tried serial mode?  You mentioned EM interference.  Have you examined your signal with an oscilloscope?

I don't have enough information yet to really understand the problem you're experiencing, much less the cause.  You say it "has been working," and that it goes "offline" for 45-60 minutes after the pump has been running for 10-15 minutes.  Did I get that right?  Can you elaborate and tell me more?

Please send me what I asked for, answer my questions, and we'll go from there.

All the best,

Bud

Surfer2010, 2021-05-30:
Hello,

thanks for your very good and detailed answer. We today exchanged the cable (LAN cable) going to the sensor and now it works fine. So there must have been some bad points/breaks inside the cable.

To answer your questions, yes i did elaborate the behaviour inside the orange tube before and it works very good. The TFminiPlus was my choice because it is waterproven and the HC-SR04 we tried before corroded very quick. Also we did try some underwater-preasure sensores but they also did not quite work.

If the project is interesting for you I can give you a more detailed and better (and complete) description of it?
So far it seems like it's working perfectly now, also no "45min break" so EM-interferrance seem to have no effect or are not present.
So I'm very happy but also glad for your quick and extensive answer. I will mark this as closed but thanks for your effort. Sorry having bothered you "to early" :-)

If you want to know more details as I said, tell me ^^

budryerson, 2021-05-31:
Sure!  Tell me everything.  I'd especially like to know how you configured your cable.
Thank you,
Bud
< bud at budryerson dot com >

closed 2021-05-31 by budryerson