budryerson/TFMini-Plus #36: Setting precision to mm

berbaba, 2021-06-09:
Hello!

I have been trying to change the precision of the device's reads but have been unsuccessful.

The TFMPlus.h file includes a line ([137](https://github.com/budryerson/TFMini-Plus/blob/master/src/TFMPlus.h#L137)) with the definition for STANDARD_FORMAT_MM assigned to a certain hexadecimal value. This is next to other commands like SET_FRAME_RATE (with its corresponding hexadecimal value) that are used with the [sendCommand() function](https://github.com/budryerson/TFMini-Plus/blob/master/src/TFMPlus.cpp#L195) in the TFMPlus.cpp, but said function works for the following commands:
SET_FRAME_RATE
SET_BAUD_RATE
OBTAIN_FIRMWARE_VERSION
SYSTEM_RESET
RESTORE_FACTORY_SETTINGS
SAVE_SETTINGS
and the change of format is not implemented here.

I would like to be able to fix this but I don't have enough knowledge and the warning of causing the device to be unresponsive prevents me from changing some things.

As far as I understand it should follow the commented steps in the function:

```C 
bool TFMPlus::sendCommand( uint32_t cmnd, uint32_t param)
{
   // Step 1 - Build the command data to send to the device

 if( cmnd == STANDARD_FORMAT_MM)     // If the command is Standard Format MM...
    {    
              memcpy( &amp;cmndData[ 3], &amp;param, X);     // add the X byte Standard Format parameter. 
// 3 is the array position for the example I copied (SET_FRAME_RATE)
// but maybe in this case is different. I don't know X either.
    }
 
     // Step 2 - Send the command data array to the device

     // Step 3 - Get command reply data back from the device.

    // Step 4 - Perform a checksum test.

    // Step 5 - Interpret different command responses.

    // Step 6 - Set READY status and go home

}
```

I assume steps 2, 4 and 6  should stay the same, but don't know if this command generates a reply or not (steps 3 and 5).
At the moment there is not much more I can do.

Anyway, thank you for the library!

budryerson, 2021-06-09:
Berbaba,

The two formatting commands do not require a parameter, so the `param` byte is set to 0.
In the included example sketch, the command would look like this:

`tfmP.sendCommand( STANDARD_FORMAT_MM, 0))`

Both formatting commands return only an echo of the command. They do not return a pass/fail bit or other useful information. Therefor, the library only examines the reply for a correct checksum byte.
```
// The library 'sendCommand( cmnd, param)' function
// defines a command (cmnd) in the the following format:
// 0x     00       00       00       00
//     one byte  command  command   reply
//     payload   number   length    length
                                                 
#define    SET_FRAME_RATE             0x00030606   // These commands each return
#define    SET_BAUD_RATE              0x00060808   // an echo of the command
#define    STANDARD_FORMAT_CM         0x01050505   //           "
#define    PIXHAWK_FORMAT             0x02050505   //           "
#define    STANDARD_FORMAT_MM         0x06050505   //           "
```

Please bear in mind that what you call "precision" is only how the data is formatted.  The accuracy (±5cm) and resolution (±5mm) of the device do not change.

I hope this is helpful.  I'm not sure of your question, so please let me know if there is anything more I can add.

Bud Ryerson
San Francisco

berbaba, 2021-06-14:
Thank you very much for reply so fast! I changed it and worked as expected, the measures had an extra digit for milimetres, just needed to change the printf as having an extra cipher wouldn't show measures with 5 digits and the unit from cm to mm. I will put this as a pull request in case you want to add it to the library.
Again thank you.

Bernat
Valencia