budryerson/TFMini-Plus #38: TFMini-Plus Neuf

BAtmos, 2021-06-28:
Hello,

I just got a brand new TFMini, I installed your library, but I got this result. I may not have tried everything, but ..

Cordialement

Bertrand
---------------------------------------------------------------
Bonjour,

Je viens de recevoir un TFMini tout neuf, j'ai installé votre bibliothèque, mais j'ai ce résultat. Je n'ai peut être pas tout essayé mais..

TFMPlus Library Example - 19MAY2021
System reset: Status: TIMEOUT
 00 00 00 00 00 00 00 00
Firmware version: Status: TIMEOUT
 00 00 00 00 00 00 00 00
Data-Frame rate: Status: TIMEOUT
 00 00 00 00 00 00 00 00
Status: HEADER
Data: 00 00 00 00 00 00 00 00 00
Status: HEADER
Data: 00 00 00 00 00 00 00 00 00
Status: HEADER
Data: 00 00 00 00 00 00 00 00 00

Cordialement
Bertrand

budryerson, 2021-06-28:
Bertrand, greetings!

It appears that you are not communicating with the TFMini at all, which is much better than some weird intermittent error
Usually the problem is either:
1) Power: Can you see the little red light inside the TFMini when power is applied?
2) Serial connection: RX > TX and TX > RX  - Do the RX/TX lines cross between the TFMini and the Arduino.

It's always something.  We'll figure it out.
Bud

BAtmos, 2021-06-28:
Bud,

Thank you for your answer, yes, before sending a message to help, I check everything several times.

By dint of getting angry, I tried to connect to USB via the WINCC_TF application.
And ..., since it works on my Arduino, curious ....
While I'm lucky enough to have a specialist, I still have a question.
For months, I have been trying TFMini, Lidar, Tof, Ultrasound, but for my Rover, I still have the problem of continuously managing distances, because even with serious elements, TFMiniPlus, I have very often distance errors which disturb my analysis of distances. Example:
1st distance 230mm, 2nd 0, 3rd 40mm. ????
An idea?
Greetings.
Bertrand

_______________________________
         Bertrand DALL'AGLIO
            Chemin de l'Orée 11
             CH-1293 Bellevue

           Tél.  +41 22 958 00 41
           Fax  +41 22 958 00 44
           Mob. +41 79 217 68 60
           email ***@***.******@***.***>

budryerson, 2021-06-29:
Bertrand,
I am happy to hear that your TFMini is now working correctly.  What was the problem?
In a stable environment, your distance readings should not vary.  Errors are uncommon.
If you wrap up **ALL** of your code (libraries and sketches), a text file of your output and some pictures of your setup into a zip file and send it to me, I will do all that I can to help.
Bud Ryerson
San Francisco

BAtmos, 2021-06-29:
Bud,

Merci beaucoup de m'aider.
C'est curieux, mais comme je l'ai dis, il a fonctionné, juste après l'avoir branché sur  WINCC_TF application!!!

L'autre question, c'était pour avoir votre avis, concernant un problème que j'ai sur plusieurs Rover (avec des capteurs (5) autres que TFMPlus) qui lorsque je teste les distances me renvoi pas toujours les bonnes valeurs. Donc le Rover s'arrête.
Il faudrait une fonction pour éliminer les valeurs trop basse.

Salutations

Bertrand

Bud,

Thank you very much for helping me.
It's odd, but like I said it worked, right after plugging it into WINCC_TF application !!!

The other question, it was to get your opinion, concerning a problem that I have on several Rover (with sensors (5 * VLX53LOX) other than TFMPlus) which when I test the distances not always give me the right values. So the Rover stops.
It would take a function to eliminate too low values.

Greetings

Bertrand

budryerson, 2021-06-29:
Bertrand,
You wrote:

> it worked, right after plugging it into WINCC_TF application

That sounds like a connection error.  You found it and fixed it.  Way to go!

The VLX53LOX device appears to require a lot of calibration and has several "ranging profiles."  As with the TFMini, you need to be careful that your sampling rate and ranging rate are compatible.  If errors continue, small discrepancies can be averaged out; large discrepancies can be ignored.

The VLX53LOX data sheet is a little short on theory-of-operation, but it looks like an interesting device.  If you want to send me one, I'd be happy to play with it and give you my opinion.  

Keep up the good work.

Bud Ryerson
San Francisco
bud-at-budryerson-dot-com

closed 2021-06-29 by budryerson