budryerson/TFMini-Plus #40: 2x (or more) devices on one board?

samuk, 2021-11-30:
Is there any example code for connecting two TFMini+ devices to a single board using this library?

budryerson, 2021-12-01:
This library is for the **Serial UART** mode of the device.  Most Arduino boards have only a single hardware serial port.  Using the Arduino **SoftwareSerial** library.  Multiple software serial ports can be implemented with and a separate device can be connected to each one.  I strongly do not recommend doing that.  I do not know of any example code for that.

I recommend using the **TFMPI2C** library and the I2C communications mode of the **TFMini-Plus LiDAR** device.  I have personally attached up to three devices at a time on the I2C bus.  I would be happy to send you that code.

Best of luck,
Bud Ryerson
San Francisco

samuk, 2021-12-01:
Ah makes sense. Thanks. Would love to see the other example code please.

budryerson, 2021-12-01:
Try this out ... after you assign and test unique and proper slave addresses for each of your LiDARs, of course.
Also, be sure you've got the latest version (1.7.0) of the TFMPI2C library installed.

[TFMP_3unit.zip](https://github.com/budryerson/TFMini-Plus/files/7636030/TFMP_3unit.zip)

samuk, 2021-12-01:
Nice one thanks. I don't have devices yet, but now I know it works I'll order some.

budryerson, 2022-03-06:
I'm not sure.  Do I need a license?  Do they need to be licensed?  Please tell me more.

samuk, 2022-03-06:
I'd prefer it if you had one. It would give confidence that you won't charge me some 'license fee' in the future.

I like https://mit-license.org/ personally.

digamesystems, 2022-04-27:
I'm using an ESP32 module on a project. It has 2 UARTs in addition to the serial port used for loading code / debugging. I was able to take Bud's library to talk to both of them. You might be able to adapt this to something like a Mega, but I haven't tried that. 

https://github.com/digamesystems/esp32_dual_lidar_example

Regards,

John.

budryerson, 2022-04-27:
Thanks for the credit, John.  "My heart soars like a hawk." - Old Lodge Skins (Chief Dan George)

digamesystems, 2022-04-27:
Credit where credit is due! It's nice work. I saw your exchange with @samuk and did the little writeup with code I had from the beginning of my project. 

I can get a picture or two of the setup if folks are interested.

budryerson, 2022-04-27:
Here's a Due, driving a stepper motor with two TFMini-S mounted back-to-back.  The SDA & SCL lines separate into a heart shaped pattern (gotta keep 'em apart!) leading from a slip ring to a small bread board.  That board also connects an optical sensor on the side of the rig to provide a position sync mark.  Is this what you were asking for?  Let me know if I can bore you with some code. :smile:

[image]

https://user-images.githubusercontent.com/7094018/165644631-be9fccc5-2fa9-43fb-8eb1-622ea5a05a8a.mp4

digamesystems, 2022-07-01:
Hi, Bud.
I took a look at the code you sent Samuk. I see that you are using the three LIDARs in 'untriggered' ('self-triggered') mode. Have you tried running in externally triggered mode? I have my setup working in that mode to save power (my application is battery powered)  but occasionally the system gets into a state where one of the sensors is no longer responding. Oddly, the data and status codes I keep receiving from a call to the `getData(..., addr)` function are the same as the last valid reading I received from the sensor. This makes it tough to tell when it's hung up. Have you any tips for detecting / recovering from this? 
Thanks!

Firestarter97, 2022-07-02:
@budryerson Do you by any chance have a schematic as to how you connected your 3 tfmini plus setup to a single arduino uno board? Thanks!

budryerson, 2022-07-03:
Dear Firestarter (Firestarter?),
Enable the I2C interface on all three devices. Assign each one a unique address.  Connect the four wires (GND, VCC, SDA & SCL) from each device in parallel and attach them to the UNO.  Keep the I2C lines short and try to keep the SCL and SDA lines separate in order to minimize cross talk.  Do you really need a schematic for that?  Let me know.
Bud

Firestarter97, 2022-07-05:
@budryerson Thanks for the explanation! I won't be needing the schematic. One more question, concerning keeping I2C lines short. Would it be fine to use 12 to 18 inches? Thanks again!

digamesystems, 2022-07-05:
@Firestarter97 I was curious about this, too and ran across this reference from TI that goes into detail on some of the issues to watch out for. https://www.ti.com/lit/an/slyt770/slyt770.pdf?ts=1656812624198 

From their doc:

 _"The best approach to resolving crosstalk is to place small series resistors on the SDA/SCL lines and to minimize the parasitic bus capacitance between the SDA and SCL traces."_

Here, they are talking about PCB design. -- For leads out in space, I might try making the SDA/SCL lines twisted pairs, each with their own ground line.
 
@budryerson what are your thoughts, Bud?

Firestarter97, 2022-08-27:
@budryerson Hello, I only recently was able to purchase the tfmini-plus lidar sensors so haven't had much time to use them. That being said I'm also trying to do a triple setup. That being said, how would I as you mentioned in a previous comment above,  "after you assign and test unique and proper slave addresses for each of your LiDARs", code wise? Thanks

Firestarter97, 2022-08-28:
@budryerson  Actually never mind. I realized you have a file called TFMPI2C_changeI2C.ino that helps search for other devices and helps change the address for those new devices. I've got a triple setup working. Thanks!

closed 2022-08-30 by budryerson

digamesystems, 2022-10-11:
Cool! Looks like you are building a “poor man’s” 2-D LIDAR?

https://www.alliedelec.com/product/sick/tim551-2050001/71940124/?gclid=Cj0KCQjw06OTBhC_ARIsAAU1yOWluG6DGaMo6JZfR-JWPU1I7lHlrSnbjw2QI00TkF1432Q53zh_R4MaAiWzEALw_wcB&gclsrc=aw.ds

(That one only goes for $2600!)

John M. Price, Ph.D.
Digame Systems<diagamesystems.com> | 88 E San Fernando St. Ste. 1511, San Jose CA, 95113 | 408.840.9871 |

[lastGasp_25]<https://user-images.githubusercontent.com/7094018/165643909-a3632cac-f1c4-4959-8fc1-f0c90a64d5c6.png>

https://user-images.githubusercontent.com/7094018/165644631-be9fccc5-2fa9-43fb-8eb1-622ea5a05a8a.mp4