budryerson/TFMini-Plus #43: How to use multiple TFMini Plus and TF02 Pro sensors on AT Mega 2560?

shivamkk007, 2022-03-12:
Hey Bud,
 I have been recently working on a project which requires multiple TFMini Plus sensors and TF2Pro sensors to be connected in a single AT Mega 2560 module. Can you tell me how can I achieve that and what should be the code for it because if I am trying it's not working. Only one value is getting transmitted and received not from the other sensor please help me on this.
Thank you!

budryerson, 2022-03-13:
Use the I2C communication mode.  Send me what you've got so far and I'll take a look at it.  Pictures are helpful.
I've had two TF02-Pro and neither of them worked, so I can't be of much help with that.
Bud

shivamkk007, 2022-03-13:
`#include <TFMPlus.h>

#include<SoftwareSerial.h>
SoftwareSerial mySerial1(50,51);
SoftwareSerial mySerial1(52,53);

float dist;
float speed1;
int dist2;          //actual distance measurements of LiDAR
int strength;
int strength2;/     /signal strength of LiDAR
float temprature;
float temprature2;
int check;
int check2;                 //save check value
int i;
int j;
int uart[9]; 
int uart2[9];                           //save data measured by LiDAR
const int HEADER=0x59;
const int HEADER2=0x59;   //frame header of data package
void setup() {
Serial.begin(115200);            //set bit rate of serial port connecting Arduino with computer
mySerial1.begin(115200);

//set bit rate of serial port connecting LiDAR with Arduino
}
void loop() {
if (mySerial1.available()) {                                //check if serial port has data input
  if(mySerial1.read() == HEADER) {                 //assess data package frame header 0x59
  uart[0]=HEADER;
  if (mySerial1.read() == HEADER) {                //assess data package frame header 0x59
    uart[1] = HEADER;
    for (i = 2; i < 9; i++) {                                  //save data in array
      uart[i] = mySerial1.read();
}
check = uart[0] + uart[1] + uart[2] + uart[3] + uart[4] + uart[5] + uart[6] + uart[7];
if (uart[8] == (check&0xff)){                                             //verify the received data as per protocol
dist = uart[2] + uart[3]*256 ;                                           //calculate distance value
strength = uart[4] + uart[5] * 256;                                //calculate signal strength value
temprature = uart[6] + uart[7] *256;                            //calculate chip temprature
temprature = temprature/8 - 256;
Serial.print("  dist = ");
Serial.println(dist/100);                                                   //output measure distance value of LiDAR
Serial.print("\t strength = ");
Serial.print(strength);                                                   //output signal strength value
                                                                                       //output chip temperature of Lidar
        }
        delay(00);
      }
    }
  }

if (mySerial1.available()) {                                //check if serial port has data input
  if(mySerial1.read() == HEADER) {                 //assess data package frame header 0x59
  uart[0]=HEADER;
  if (mySerial1.read() == HEADER) {                //assess data package frame header 0x59
    uart[1] = HEADER;
    for (i = 2; i < 9; i++) {                                  //save data in array
      uart[i] = mySerial1.read();
}
check2 = uart2[0] + uart2[1] + uart2[2] + uart2[3] + uart2[4] + uart2[5] + uart2[6] + uart2[7];
if (uart2[8] == (check2&0xff)){                                             //verify the received data as per protocol
dist2 = uart2[2] + uart2[3]*256 ;                                           //calculate distance value
strength2 = uart2[4] + uart2[5] * 256;                                //calculate signal strength value
temprature2 = uart2[6] + uart2[7] *256;                            //calculate chip temprature
temprature = temprature2/8 - 256;
Serial.print("  dist = ");
Serial.println(dist2/100);                                                   //output measure distance value of LiDAR
Serial.print("\t strength = ");
Serial.print(strength2);                                                   //output signal strength value
                                                                                       //output chip temperature of Lidar
        }
        delay(00);
      }
    }
  }
}

`
This is the code I modified to use the TFMini Plus and TF02 Pro but the function only works when I use only one device i.e. with only SoftwareSerial mySerial1(50,51);

I am currently working on a project which uses 4 sensors, is it feasible to do so with a single AT Mega?
Looking forward for your reply.

Using I2C will highly effect the Performance I'd be expecting i.e. the refresh rate would drop from 1000Hz to 100Hz.
Is there any solution for keeping the performance intact?

budryerson, 2022-03-13:
"the function only works when I use only one device i.e. with only SoftwareSerial mySerial1(50,51);"
In your code, I only see `MySerial1()`.  I don't see `MySerial2()`.

"is it feasible to do so with a single AT Mega?"
No, it is not feasible to use UART and "a single AT Mega."  Several AT Megas connected together might work, but why?  It is _so easy_ if you use I2C.

"Using I2C... the refresh rate would drop from 1000Hz to 100Hz"
I don't think that's true.

shivamkk007, 2022-03-15:
Hey Bud, 
I have sent you an e-mail regarding same issue and an offer for being a part of a startup project. 
The e-mail is from  - kshivam523@gmail.com
Please give it a look. Looking forward for your reply.
Regards

budryerson, 2022-03-15:
I'm happy to learn that you have resolved the problem.  Thank you for allowing me to help.
Sincerely,
Bud Ryerson
San Francisco

closed 2022-03-15 by budryerson