budryerson/TFMini-Plus #49: Temperature

Ardusmith, 2023-05-28:
I'm using your library and it works great. I do receive temperature readings in 40-45 degree range. The TFMPlus returns Celsius degrees, so I think this is not correct. (My room temp is about 22ºF). So I reversed the algorithm for calculating temp:

original from library is:
temp = ( temp >> 3) - 256;

my reverse calculation:
float newtemp = ((temp+256) << 3)/100.;

(temp of 45 yields 2408 in reverse, so dividing by 100 gives 24.08... Just guessing... maybe it reports in 1/100ths?)

This gives reasonable temperatures for my physical space. Am I on the wrong track? Could benewake have updated the device to report temperature values in a different format than previous versions of the device?

I plan on using my device outside, so I would like to use the temperature data. (single digit accuracy is completely adequate)

Thanks.

budryerson, 2023-05-29:
Page 13 of the TFMini-Plus product manual (v4) gives the following formula:

> **Temp(Temperature):** Represents the chip temperature of TFmini Plus. Degree centigrade = Temp / 8 -256

Please keep in mind that the temperature being measured is the temperature of the chip inside the case, which is due mainly to the computational load and current draw.  This is not the right device to measure ambient temperature.

Ardusmith, 2023-05-29:
All good--I was just a bit surprised that the chip ran that hot. I will use a different and proper sensor for ambient temperature. Thank you very much for your prompt reply.

closed 2023-05-29 by Ardusmith