budryerson/TFMini-Plus #52: Multiple sensors

bobkeanu80, 2024-11-02:
I'm sorry for the possibly stupid question. I'm new here. Is it possible to see a sample code for working with multiple sensors. 
I want to use all 4 Arduino Mega serial ports

digamesystems, 2024-11-02:
Hi Bob.
I posted this a while back. https://github.com/digamesystems/esp32_dual_lidar_example
It's written for an esp32, but a lot should map to the Arduino. I hope it helps. 
Regards,
John Price

budryerson, 2024-11-02:
Bob - Other than the programming port (UART0) there are only three additional serial ports on the Mega 2560.  That said, it is cleaner, easier and just better to use the I2C interface to run multiple Lidars. If you _**must**_ use serial, please let me know.

[image]

digamesystems, 2024-11-02:
I2C is a nice way to go. I made an I2C example as well: https://github.com/digamesystems/DualLIDARI2C

bobkeanu80, 2024-11-03:
> Hi Bob. I posted this a while back. https://github.com/digamesystems/esp32_dual_lidar_example It's written for an esp32, but a lot should map to the Arduino. I hope it helps. Regards, John Price

Hello, John! Thank you for your prompt response! This is a very useful example, thank you!

bobkeanu80, 2024-11-03:
> Bob - Other than the programming port (UART0) there are only three additional serial ports on the Mega 2560. That said, it is cleaner, easier and just better to use the I2C interface to run multiple Lidars. If you _**must**_ use serial, please let me know.
> 
Hello, Bud! Thank you for the prompt response!
As far as I know, the Arduino Mega has only one I2C port, and I would like to solve my task with one Arduino and i need four TF Luna, so I'm considering using 4 serial ports.

bobkeanu80, 2024-11-03:
> As far as I know, the Arduino Mega has only one I2C port, and I would like to solve my task with one Arduino and i need four TF Luna, so I'm considering using 4 serial ports.

Thank you, John! As far as I know, the Arduino Mega has only one I2C port, and I would like to solve my task with one Arduino and i need four TF Luna, so I'm considering using 4 serial ports.

budryerson, 2024-11-03:
Bob - Whereas one UART port connects to only one single device, you can connect up to 112 separate devices on a single, two-wire I2C port. Be aware that the I2C protocol was designed for short distances, and be careful about cross talk between the signal wires.  Try it out.  You'll have fun.  Otherwise, I think you're looking at purchasing a serial port multiplexer because, as I said, you only have **_three_** additional serial ports on the Mega 2560.  Let me know what you decide.
All the best,
Bud