budryerson/TFMini-Plus #54: Compatibility with TFS20-L Benewake Lidar

bobvogt314159, 2025-03-07:
I have been trying to use the TFMini-Plus Library with a TFS20-L model.  I get "TIMEOUT" messages and suspect an incompatibility after implementing at an array of wiring variations.  I just got ahold of the user guides for both models (TFMini-Plus and TFS20-L) but have not dug through them to check for differences.  I will do this but thought I would ask the simple question.  I intend on using the serial port communication protocol for my project, with serial port #2 on an Arduino Giga R1 board.

Thanks in advance for your hep.

Bob

digamesystems, 2025-03-07:
Hi Bob. 

You should be able to make that work. I did a little example some time ago to use two serial ports on an ESP32 dev board to talk to two separate TFMini-Pluses. (I was using them for direction of travel detection) Different processor, but the code should map over to the Giga, I believe. 

https://github.com/digamesystems/esp32_dual_lidar_example

You also might want to look into Bud's I2C interface as well. It's quite snappy. A similar example is here: 

https://github.com/digamesystems/DualLIDARI2C

Hope this helps. 

Regards,
John

bobvogt314159, 2025-03-07:
Hi John,

Thanks for your reply and links.  I am still somewhat concerned with compatibility and the TFS20-L, but will keep on digging.

Bob

digamesystems, 2025-03-07:
Hi Bob.

No problem.

The TFs20-L looks like a nifty little sensor. (Nice range!)

I found these:

Datasheet: Benewake_TFS20_L_Datasheet-3539498.pdf<https://www.mouser.com/datasheet/2/1099/Benewake_TFS20_L_Datasheet-3539498.pdf>
Benewake TFS20-L - Mouser Electronics<https://www.mouser.com/datasheet/2/1099/Benewake_TFS20_L_Datasheet-3539498.pdf>
1 ©2024Benewake(Beijing)Co.,Ltd.·Allrightsreserved·Subjecttochangewithoutnotice TFS20-L is a single-point LiDAR module of Benewake. Multiple built-in operating modes let
www.mouser.com
Users Manual w/ Protocol description: Benewake_09-20-2024_TFS20-L.pdf<https://www.mouser.com/catalog/specsheets/Benewake_09-20-2024_TFS20-L.pdf>
TFS20 L User Manual - Mouser Electronics<https://www.mouser.com/catalog/specsheets/Benewake_09-20-2024_TFS20-L.pdf>
Contents 1 Product Overview ..... 1
www.mouser.com